AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #130: New feature idea

itsDaniel-23, 2021-02-21:
Another very great feature would be to let the battery charge slower to the preferred value. Let’s say I used my battery all day and would like to charge it over night to 70%. If there’s a way to charge it slower to the preferred value (because it has all night to get to that value) it would be my guess that its beneficial for the overall health of the battery. 
I’m not certain if that’s possible or even going to have an impact on battery health. Its just an idea that I came up with recently.

TheMazus, 2021-02-21:
Hi itsDaniel-23,
Thank you for the idea, we have also thought about this feature. Unfortunately, we don't know if this is even possible.

LeshawnG, 2021-02-21:
It's a great idea!
Would this be something like "using a 5W instead of the included 18W with the iPhone Pro" to conserve battery health as charging at higher voltage and amperage wares the battery quicker? I've heard this quite a bit before but I need to do more research. Higher wattage causes more heat though which we know for certain is bad for a battery.

My workaround before AlDente was released for M1 was using the 18W Apple charger instead of the included 61W for the MBP. The 61W supplies 20V/3A and the 18W supplies 9V/2A. Check with `pmset -g ac`.
I would assume it's safe to charge at a lower voltage/amperage since for iPhones you can use either the 18W fast charger or the 5W. The real question to answer is what difference, if any, does lower power input make to battery chemistry over time. I'll do some more research. 
I'll probably stick with using the 18W for my MBP as I hardly use it for my iPhone and I prefer slower charging for the iPhone with 5W.

LeshawnG, 2021-02-21:
Also @TheMazus @davidwernhart another feature I think would be nice:
An option to manually stop charging. It's a simple idea and can function as one button that changes text based on the action it would take when clicked. 
It's already easy to do this by just setting a lower charge level than battery level but I think a button to manually stop charging would be a neat addition. 
But this means they still have to move the slider and enable allow discharging if they want a lower level and charging would be automatically inhibited.
I'm trying to think about how useful this would really be. I guess above is one case where it's useless but I still think it's neat to have a button that stops charging!

TheMazus, 2021-02-21:
> It's a great idea!
> Would this be something like "using a 5W instead of the included 18W with the iPhone Pro" to conserve battery health as charging at higher voltage and amperage wares the battery [...]

Hi LeshawnG,
I need to do more research in this area too but it is commonly thought that charging with higher power results in faster battery degradation. This is also the reason, why Tesla limits the maximum power at superchargers when people have charged at superchargers very often.

TheMazus, 2021-02-21:
> Also @TheMazus @davidwernhart another feature I think would be nice:
> An option to manually stop charging. It's a simple idea and can function as one button that changes text based on the action it [...]

We have also thought about something like this but we are not sure how to implement it. Maybe just a little button to stop and start charging would be the most convenient solution.

jakuSk, 2021-03-19:
> @LeshawnG
> 
> > My workaround before AlDente was released for M1 was using the 18W Apple charger instead of the included 61W for the MBP. The 61W supplies 20V/3A and the 18W supplies 9V/2A.
> 
> I use a 5W for my 12 Pro as well. Plenty fast! But 18W to 5W isn't as big a decrease as 61W to 18W. Are you sure the 18W adapter is able to provide necessary power to run the Macbook without it needing to rely back to 'splicing in' additional power from the battery? Also, I imagine it literally taking forever to charge, right?

> Another very great feature would be to let the battery charge slower to the preferred value. Let’s say I used my battery all day and would like to charge it over night to 70%. If there’s a way to [...]

Literary 2nd sentence in this post has an answer for you my friend.

MatthiasKerbl, 2025-02-06:
Hi Everyone,

We already have an issue/feature request open at #426. We will give updates regarding this topic over there.

Best,
Matthias

closed 2025-02-06 by MatthiasKerbl