AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #137: High CPU Consumption

liysky, 2021-02-22:
Macbook Pro 2017 without tourch bar.
Mac OS 11.2.1
sometimes with the high CPU consumption and fan with high speed and noise.
[image]

B4nanaBean, 2021-02-22:
I have a similar issue on M1 MBA. It's nowhere severe as yours, but the CPU consumption gradually goes up as the time goes on. AlDente seems to consume quite high resources for what it does.

hansibansix, 2021-02-22:
same on macbook air m1, cpu consumption starts with 0.5% and gradually rises up to 30%.

leeummm, 2021-02-23:
Agreed. CPU and energy consumption seems high. It appears to be worse when under your charge limit, as if it is constantly checking to see if you have hit your charge limit. Unsure if that is true.

Other thing that is clear: the longer it's running, the more energy and CPU it uses.

B4nanaBean, 2021-02-23:
Ok, so I've been running the program for straight 12 hours today, and it shows a different behavior. Yesterday the CPU usage went up. Today it stays ~0.3% usage (update frequency 5sec) throughout the whole day. I don't know why it is acting completely different. Just a note, I did turn off the computer last night and boot up in the morning.

leeummm, 2021-02-23:
> Ok, so I've been running the program for straight 12 hours today, and it shows a different behavior. Yesterday the CPU usage went up. Today it stays ~0.3% usage (update frequency 5sec) throughout the whole day. I don't know why it is acting differently. Just a note, I did turn off the computer last night and boot up in the morning.

Have you been plugged in all day? At or below your desired charge level?

How has energy usage trended?

B4nanaBean, 2021-02-23:
[image]

Exactly at the desired charge level.

leeummm, 2021-02-23:
> [image]
> 
> Exactly at the desired charge level.

Interesting. My anecdotal experience has been that energy use stays low when you are both plugged in and at your desired charge level.

B4nanaBean, 2021-02-23:
For the energy usage trend, I don't know since I didn't monitor it.

B4nanaBean, 2021-02-23:
M1 Mac fellas, I have a theory for establishing a stable cpu consumption of AlDente. It seems that if you give AlDente a good 3 minutes BEFORE you plug in the charger, it uses low resources. If you were already running AlDente with high cpu usage, quit the program and restart it. If you see somewhere between 0.1%-0.3% cpu usage after plugging in the charger, I think you are good.

qfettes, 2021-02-23:
M1 Macbook Pro 13: the cpu usage and total cpu time seems abnormally high for me; see image. When actually opening the application, CPU% can jump above 10% quickly. Way too high for what it is supposed to do.

[image]

LeshawnG, 2021-02-23:
AlDente lingers around 5-7% CPU use for me and it does take longer to open the longer it's running.
About to power off and wait a while before plugging in the charger as mentioned above.
M1 MBP 13
[image]

Tim675, 2021-02-23:
I seem to be unfortunately having the same issue as others. The longer the app is open the CPU usage slowly creeps upwards unless I restart the app which seems to temporarily reset it.

leeummm, 2021-02-24:
I haven't tested this enough but I am convinced the issue has to do with AlDente checking if you have hit the charge limit...

Limeybastid, 2021-02-24:
Not bad slight charging occurs as can be see on the attached, but that is to be expected since my MBP is off charge sometimes and in standby. CPU usage seems good.
[image]
[image]
[image]

MasterOfMouse, 2021-02-24:
Are you on 2.0 or 2.1?

Tim675, 2021-02-24:
I am on 2.1 and having the issue. Interestingly when I enabled "allow discharge" the (feature in beta) the cpu usage hasn't yet began to creep up as of yet.

liysky, 2021-02-24:
When I use the big application like Parallels Desktop, Photoshop and so on, CPU usage of it start to creep up. I wonder if there's some reason for this issue.

MasterOfMouse, 2021-02-24:
> I am on 2.1 and having the issue. Interestingly when I enabled "allow discharge" the (feature in beta) the cpu usage hasn't yet began to creep up as of yet.

I'm on 2.0 by the way. But I doubt its something in 2.1.

MasterOfMouse, 2021-02-25:
I have updated to 2.1. I will update on the CPU usage.

Tim675, 2021-02-25:
Ive been running with the option to "allow discharge" enabled now for a few days now and my CPU use has remained low. Maybe a bug that occurred when enabling the new feature? I'll just leave it on for now. I'm on 2.1 BTW.

MasterOfMouse, 2021-02-26:
I have used 2.1 for two days now CPU usage is usually between 0.2 and 0.5. The energy over 12 hours is 0.43. I guess its fine for now

lpla, 2021-02-26:
@bobouc @B4nanaBean @hansibansix @leeummm @qfettes @LeshawnG , in which version of AlDente are you having this CPU usage issue? 2.0 or 2.1?

bobouc, 2021-02-26:
2.1

qfettes, 2021-02-26:
> @bobouc @B4nanaBean @hansibansix @leeummm @qfettes @LeshawnG , in which version of AlDente are you having this CPU usage issue? 2.0 or 2.1?

I'm also on 2.1

leeummm, 2021-02-27:
It was present on 2.0 and 2.1 for me.

Tim675, 2021-02-28:
Just to offer an update CPU usage has begun to increase again as well with "Allow discharge enabled". Before it had only increased when this was disabled for me. I'm on 2.1 here.

patperek, 2021-02-28:
M1 MBP, my CPU usage seems to jump between 0 and 15.2% every 2 seconds in Activity Monitor. If the battery management itself is supposed to be handled by the SMC (or whatever the equivalent is in M1 Macs), then what could explain these CPU jumps? I don't see what else it should be doing after writing to the SMC, does anyone have any idea?

qfettes, 2021-02-28:
It's probably somewhat similar to what @leeummm suggested; rather than sleeping, the application is probably spinning silently and eating a lot of cpu time.

patperek, 2021-03-02:
I've started setting the charge limit, waiting for it to reach the limit and start running off AC, and quitting Al Dente (using the Quit button in the app). The charge limit stays and the CPU consumption is a non issue as the app is closed. Could there be any issues with this approach while the CPU usage issue isn't fixed?

patperek, 2021-03-02:
Additionally, restarting the app 5 minutes after closing it and my CPU usage has gone down from 10-15% to 0.4% while idling, on my M1 Macbook Pro. The charge is still being limited as desired.
[image]

bobouc, 2021-03-02:
> Additionally, restarting the app 5 minutes after closing it and my CPU usage has gone down from 10-15% to 0.4% while idling, on my M1 Macbook Pro. The charge is still being limited as desired.
> [image]

> I've started setting the charge limit, waiting for it to reach the limit and start running off AC, and quitting Al Dente (using the Quit button in the app). The charge limit stays and the CPU [...]

You are right. Quitting the AIDente(using the Quit button in the app) is a better way for resolving "High CPU Consumption". I do that like you mentioned above, and it works ( still has the charge limit ).

qfettes, 2021-03-03:
If the app can be turned off and the full functionality is maintained, what could it possibly be doing in the background to use so much CPU?

maiksd, 2021-03-05:
Same here with 2.1 on an MBP 15" 2018. Just noticed that AlDente ate 9 hours of CPU time since I last rebooted 7 days ago...

qfettes, 2021-03-07:
I don't think this should be closed. There wasn't really a resolution; there was only somewhat of a workaround.

Tim675, 2021-03-07:
Yea issue is ongoing. Definitely not closed.

bobouc, 2021-03-07:
Reopen now.

TheMazus, 2021-03-07:
Hi Guys,

Thank you very much for observing this issue in such detail. @davidwernhart and I are aware of it and working on fixing it. However, we are not completely sure what the cause of the high CPU consumption is since we currently cannot replicate the issue on our systems.

Best regards,
Matthias and David

Sixfifty, 2021-03-08:
I'm getting the same issue on a MacBook Pro 13" M1 with v2.1. Never installed v2.0.

Quitting and re-opening the app every few days seems to solve the issue, but ideally you shouldn't have to quit at all.

maiksd, 2021-03-12:
For me the CPU consumption stopped after updating AlDente to 2.1.2.

firru, 2021-03-12:
> I had the same problem with version version 2.1 on MacOS 11.2.2.
> I tried using the "classic smc" option but this had only a minor effect reducing the cpu usage to 2 to 7 %.
> 
> After updating to 11.2.3 the app shows (when running in background) a cpu usage of 0.x %.
> 
> I hope this helps solving this issue. Maybe other users can share their impressions after updating to 11.2.3.

I was to enthusiastic: Later today the cpu usage is again at about 6 %.
I will try to update to 2.1.2.

trr-amsiq, 2021-03-12:
Seeing same issue on 2018 15" MacBook Pro i9 running MacOS 11.2.3 with AlDente 2.1.2

[image]

SunDiego1, 2021-03-14:
> Macbook Pro 2017 without tourch bar.
> Mac OS 11.2.1
> sometimes with the high CPU consumption and fan with high speed and noise.
> [image]

I have the same issue with AlDente 2.1.2, as a temp fix, I reverted back to AlDente 1.2 and it fixed the issue.

qfettes, 2021-03-16:
I just wanted to jump in and update on my M1 Macbook Pro. Usually I just turn off AlDente while I'm not charging to avoid this issue. I wanted to test and see if I could still observe the issue, so I let the app run for 24 hours consecutively. When first starting, CPU usage is reasonable (usually 0.x%). After 24 hours CPU usage bounces between 6% and 20% (!!) with over 31 hours of CPU time over the last 12 hours (more than my web browser in the same period).

osianSmith, 2021-03-16:
I've run into this on my MacBook pro 2019. Seems to be a relaunch is enough to fix it. It was open for about 5 days before I notice it happened.

It was a high single-threaded load. My instinct tells me its an infinitive loop that is linking back to recursive data? I haven't been able to try to debug it myself though

vassudanagunta, 2021-03-17:
@MatthiasKerbl 

> However, we are not completely sure what the cause of the high CPU consumption is since we currently cannot replicate the issue on our systems.
> 
> Best regards,
> Matthias and David

Even if you don't se CPU consumption over a few percent, does AlDente rank as one of the top consumers on your systems? As you can see on mine it is in the top 5, ranking above and using many times more CPU than Safari, mds and even DropBox which is a non-native Python client constantly monitoring file system events.

[image]

firru, 2021-03-18:
> I was to enthusiastic: Later today the cpu usage is again at about 6 %.
> I will try to update to 2.1.2.

2.1.2 has the same issue.
But I have _more/new_ infos now:

_1. I had the high cpu usage.
2. I quit the app and restarted it.
3. No high cpu usage.
4. I opened the app dialog and opened the settings (and closed it).
5. I did other things and after a while I had the higher cpu usage again._

**It seems that the app only has higher cpu usage if there was in interaction with the app dialogs. If the app is (re)started and no interaction with the app dialogs there is no higher cpu consumption.**

bobouc, 2021-03-18:
> > > I had the same problem with version version 2.1 on MacOS 11.2.2.
> 
> > > I tried using the "classic smc" option but this had only a minor effect reducing the cpu usage to 2 to 7 %.
> 
> > > After updating to 11.2.3 the app shows (when running in background) a cpu usage of 0.x %.
> 
> > > I hope this helps solving this issue. Maybe other users can share their impressions after updating to 11.2.3.
> 
> > 
> 
> > I was to enthusiastic: Later today the cpu usage is again at about 6 %.
> 
> > I will try to update to 2.1.2.
> 
> 
> 
> 2.1.2 has the same issue.
> 
> But I have _more/new_ infos now:
> 
> 
> 
> _1. I had the high cpu usage.
> 
> 2. I quit the app and restarted it.
> 
> 3. No high cpu usage (about 35 min. cpu time for a whole work day).
> 
> 4. I opened the app dialog and opened the settings (and closed it).
> 
> 5. I did other things and after a while I had the higher cpu usage again._
> 
> 
> 
> **It seems that the app only has higher cpu usage if there was in interaction with the app dialogs. If the app is (re)started and no interaction with the app dialogs there is no higher cpu consumption.**

The issue can be easily reproduced(in my own try). And the developer should pay attention to this heavy issue that occurred in many many people's mac. Obviously this issue has become the most discussed issue on issues list. In my view, it should be on the first priority.

davidwernhart, 2021-03-18:
Hi guys,

Thanks for keeping us updated on your findings on this issue. We have also made some progress and did some profiling. The issue seems to come from SwiftUI taking up more and more CPU resources to redraw the view as time goes on. Sadly, this does not make any sense to us, since the rate at which AlDente updates the variables that are relevant for the view remains constant. To make it even harder to debug, this issue does not occur in every system, which is even more puzzling to us. Any help and extra info regarding this is appreciated!
Nevertheless, we think we might have found a way to avoid this issue, but we need more time to implement and test it properly. We will keep you updated and release the fixed version as soon as possible.

Best regards,
David and Matthias

leeummm, 2021-03-18:
> 2.1.2 has the same issue.
> But I have _more/new_ infos now:
>
> _1. I had the high cpu usage. 2. I quit the app and restarted it. 3. No high cpu usage (about 35 min. cpu time for a whole work day). 4. I [...]

I also believe the experience is the same on my side. If I don't ever open the menu bar view, the CPU consumptions stays close to 0.

leechaojie, 2021-03-23:
Same problem.
CPU usage rises after booting for a while, and over time it gets higher and higher and has to quit and restart.
Interestingly, versions prior to 2.1.2 did not encounter this problem.
MacBook Pro (16-inch, 2019)

MatthiasKerbl, 2021-03-24:
Hi Guys,

we have finally managed to fix the high CPU consumption issue with our new version AlDente 2.2 which was just released. We also added an update function for future updates.

Hope you like it,
David & Matthias

Tim675, 2021-03-25:
CPU now running at 0.0 percent for me and holding steady. Great work! Very happy. Can't wait for pro version!

firru, 2021-03-26:
> CPU now running at 0.0 percent for me and holding steady. Great work! Very happy. Can't wait for pro version!

Same for me! After updating 0.0 % and very low cpu time (as expected).
Great job!

bobouc, 2021-03-26:
Awesome Man! Great work for the latest version 2.2.

closed 2021-03-26 by bobouc