AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #138: Use with caution: Ruined my MBP 15" Late 2013 battery

svintuss, 2021-02-24:
I'm sorry to say but AlDente screwed my MBP's battery.

I've got mine replaced in early 2018 just before Apple ceased Late 2013 models support. Up until installing AlDente the battery had 174 cycles and 92%-94% health. 

After using AlDente for about a month on 65% setting my laptop simply shuts down when charge level drops below 47%. No hibernate, nothing. 47% > Low power alert and 1% charge > again 47% after power on. Cycled the battery a couple of times, no improvement so far.

LeshawnG, 2021-02-24:
I'd really like to believe that one month using AlDente could do something like that to a MB battery. I'd highly doubt it can be anything to do with the physical state of the battery as keeping it at one level for a month (65%) is not abnormal.

Can you give some more info about your MBP? Did you purchase your 2013 MBP secondhand/preowned? Also can you give a timeframe for how long it took to accumulate those 174 cycles?

svintuss, 2021-02-24:
Me too I was sceptical at first that keeping the battery at 65% would do any harm. But I find it very suspicious that after the only thing that changed in my workflow being AlDente a perfectly fine battery  became unusable.

I've had the laptop since new. In 2018 replaced the battery (with the entire top case) in Apple's original service workshop. 
The battery manufacture date is 2018-03-06 and it was installed in Dec. 2018. Since then it has gone 174 cycles (I mostly use my MBP plugged in).

davidwernhart, 2021-02-24:
Hey @svintuss 
sorry to hear about that.

Of course, I cannot definitely rule out AlDente as being the cause of your problems.
But since the app's first release about a year ago, more than 20000 people have downloaded it and I strongly suspect that issues like this one would be far more common if it had anything to do with AlDente.
The tool only utilizes Apple's built-in battery control functions that are also utilized by the MacOS "optimized battery charging" feature for example.
Also, battery maintenance and cell chemistry is a very complex topic, and AlDente cannot save your battery from degrading over time. I would rather compare it to eating healthy to reduce the risk of getting cancer or other serious illnesses at some point, but sadly there is never a guarantee for everything.

Best regards,
David

svintuss, 2021-02-24:
@davidwernhart thanks for reply. 
I sure understand that batteries are degrading over time and stuff, but one has also bear in mind that Apple opted out old systems from "optimised battery charging" for a reason. 

P. S. Now that I've fully charged the battery it's stuck at 100% for 30+ minutes. Definitely something went loose in calibration.

timension, 2021-02-24:
> I'm sorry to say but AlDente screwed my MBP's battery.
>
> I've got mine replaced in early 2018 just before Apple ceased Late 2013 models support. Up until installing AlDente the battery had 174 cycles [...]

I had a simliar issue and AlDente didn't even exist then. The problem was exactly what AlDente addresses - my MBP run mostly on AC power and was fully charged most of the time. I had very few cycles and the battery health showed nearly 100 %.

One of a sudden during one of the rare uses when not connected to AC my MB just died at 40 % or something. It could start again in order just to die again within a couple of minutes. The problem gradually increased. The MB started dying at 50%, 60% and so on, until it reached 95% and virtually couldn't run on battery at all.

Eventually I brought it to Apple where they could conclude that my battery was ruined. The probable cause was the constant AC supply. They showed me the battery health graph that was almost horisontal but at the end it just collapsed down. They also compared to a normal graph, where the line gradually declines.

To sum up. Most probably you have the same issue (caused by constantly high SoC or not) and installing AlDente was just an unfortunate coincidence.

TheMazus, 2021-02-24:
> I had a simliar issue and AlDente didn't even exist then. The problem was exactly what AlDente addresses - my MBP run mostly on AC power and was fully charged most of the time. I had very few cycles [...]

Hi @timension,
Thank you for the great summary. That's exactly what I think happened with @svintuss's battery and I had the same experience with one of my MacBooks. I had it plugged in for roughly 1,5 years with minimal degradation but in one week it was completely broken. The Capacity dropped significantly and it always shut down with 50% left.
Currently, I am using a 2016 15" MacBook Pro for 7 months now with AlDente and the capacity dropped only 0.7% which is in the margin of error.

TheMazus, 2021-02-24:
> I'm sorry to say but AlDente screwed my MBP's battery.
>
> I've got mine replaced in early 2018 just before Apple ceased Late 2013 models support. Up until installing AlDente the battery had 174 cycles [...]

Hi @svintuss,
I am sorry about what happened to your MacBooks battery but as everyone else said, it is highly unlikely that AlDente has anything to do with it. I hope you give AlDente a second chance with a potential new battery to extend its life.

closed 2021-02-24 by TheMazus

svintuss, 2021-02-25:
Ok guys, I feel like I need to write a follow-up. 

After once again charging to 100% and resetting SMC my MacBook successfully went into hibernation at 1% charge. I plugged it in and charged it to 100% afterwards and it even displays the same battery capacity as before (ca. -1% which is tolerable). 

I still stand by my point that AlDente is enabling a usage scenario which was unintended by Apple (at least for those old pre-2016 systems). In my case a month of staying at 65% was enough to distort battery calibration values in such a way that two full charge-discharge cycles were needed to return to normality.

So yes, it's a conceptually great tool but use it wisely.

qfettes, 2021-02-25:
I think the commenters on the thread want see evidence other than "aldente is definitely the culprit for calibration issues on my old macbook." Old batteries can have calibration issues and doing a full power cycle might help, whatever the cause of the issue.

I'm not a contributor to the project and I want to be pragramatic about it so I can understand the tool I'm using. Keep us updated if you use aldente in the future and calibration issues resurface, or if you never use it again and the battery has no further issues. Perhaps the devs here would be able to point you to some relevant logging information concerning battery capacity/usage/health that you could share?

TheMazus, 2021-02-25:
> Ok guys, I feel like I need to write a follow-up.
>
> After once again charging to 100% and resetting SMC my MacBook successfully went into hibernation at 1% charge. I plugged it in and charged it to [...]

Hi @svintuss,

Thanks for the update. Awesome to hear that your battery is not permanently damaged because of degradation. It really looks like calibration was the problem and AlDente might have contributed to a faulty calibration. Since there is some evidence that keeping Li-Ion batteries at lower percentages will not only result in a slower degradation but can also result in a disturbed calibration when no full cycles are done over a longer period. Therefore, most people recommend doing a full cycle at least once a month. I will definitely update our readme, to let people know about this.

Thanks again for sharing!

Best regards,
Matthias

giovannitro96, 2023-03-18:
I bought a new Macbook Air m1 2020 and I've always used it as a developer to build ms and study for university, since last month I had 99% of battery maximum capacity. Last month I saw a video where someone was talking about battery corruption with Apple's setting "optimised battery charging", he suggested to use AlDente to avoid this issue and be safe. 
After 1 month of usage of AlDente, my maximum capacity is 93%.

lampshaj, 2023-06-20:
I'll jump onto the wagon. My 2018 Macbook Pro's battery started crapping out in 2022 after using Al Dente for a few months. It seemed way out of calibration because the computer would suddenly turn off at around 40% charge.