AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #139: App compatibility with Mac OS Mojave or Catalina

yiqiangjizhang, 2021-02-24:
Hi! This app is what I was searching for years! And I finally saw it, however, I am using a Macbook Pro 2017 with Mojave and I don´t want to upgrade to Big Sur since there was a ton of problems. I wanted to ask if the developers will make a version compatible with Mojave or Catalina.

Thank you all! I will definitely buy the pro version if they make it compatible with Mojave!

svintuss, 2021-02-24:
You could just download v1.2, it works under Catalina. 

However Mojave support would be nice too (pre-2013 systems won't go higher than 10.14).

yiqiangjizhang, 2021-02-24:
I´ll just upgrade to Catalina then just for this App! Jajajaajja cause I love the idea!

closed 2021-05-17 by yiqiangjizhang