AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #142: Charge Limit doesn't reset when quitting AlDente

qfettes, 2021-02-25:
Device: M1 Macbook Pro 13"
Steps to reproduce: 

1. Set max charge <= to current battery level
2. Plug in laptop. It should be on plug power but not charging
3. Reset max charge to 100
4. Unplug Laptop
5. Quit Al Dente
6. Plug in laptop. It will be powered by the plug but not charging. 

I don't want to run Al Dente all of the time because it is kind of a resource hog (see: #137). But this bug forces me to use it every time I want to charge my laptop.

qfettes, 2021-02-25:
Maybe this is known and low priority. If I wait a few seconds 5-20 for the laptop to start actively charging before quitting aldente, this doesn't happen. It's still a minor inconvenience.

TheMazus, 2021-02-25:
Hi @qfettes,

I am not able to reproduce your bug on my Intel Macbook even if I do all those steps really fast. Maybe it is just a problem on Apple Silicon Macs. However, you should always wait up to 2 minutes for any changes from AlDente to take place.

Best regards,
Matthias

closed 2021-04-22 by MatthiasKerbl