AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #145: After 12.2.2: M1 won't wake up from sleep in clamshell mode, even if connected to power

vividfog, 2021-02-26:
The few-day-old Alpha version worked great for some days, but after installing 12.2.2 yesterday, I could no longer use my M1 Air in clamshell mode, while connected to a TB3 monitor ... unless the battery level was below 80%, my max setting. The Mac would not wake up when pressing an external keyboard. Tried Bluetooth and wired keyboards.

Also, there were 3 kernel panics during today. Don't know if that's related or not.

Anyway, I had to uninstall AlDente and its support software, because otherwise the Mac just wouldn't wake up when the lid was closed. It seems that the Mac thought "no charging, must sleep, lid closed, won't listen".

The setting "prevent the computer from sleeping while display off" is checked in power settings, but it is in effect only while on power. With AlDente installed, the Mac thinks it's not on power (if the charging limit has been reached). That's, how it appears to me.

What's strange is that I tried to set the charging level to 100 after the fact, that didn't help. But I can say that I could use the computer from 78% up to 80%, before it fell to sleep and didn't wake up, until I opened the lid, or until I uninstalled AlDente.

I also tried to reinstall the helper tools to no avail.

Suggestions?

vividfog, 2021-02-27:
I tried to install AdDente again, to isolate what's happening. I can run the app in clamshell mode, I can set the charge limit. But if I turn on "Allow Discharge (beta)", the Mac immediately goes to sleep and does not wake up, not until I open the lid. This setting was previously on for me and I didn't notice any problems until yesterday. Can anyone else test this behavior on 12.2.2?

The charge limit is now set to 80. I'll unplug the Mac to drain the battery a little and will observe how things work. Won't be using the "Allow Discharge" feature then. I know the feature isn't supposed to work in clamshell mode, but it is unexpected to see the whole Mac not work when in clamshell mode, while this feature is enabled.

davidwernhart, 2021-02-27:
#114

closed 2021-02-27 by davidwernhart