AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #153: strange switch-on and shut-down on MBP retina 15-inch, mid 2014 (AlDente v2.0-alpha)

andreafurla, 2021-03-01:
Hi there,
when switch-on my Mac, after login and when it finish to load all, the Mac goes into standby.
When shut-down it has an error, and restart itself immediately.

These behaviors happen at first time.
At second time all it's OK.

TheMazus, 2021-03-01:
Hi @andreafurla,
you should install our latest version 2.1-alpha. Maybe this will solve your issues.
Best regards,
Matthias

andreafurla, 2021-03-01:
Hi @TheMazus ,
Have I to uninstall v2.0? or simply download and install the last version?
Thank you

TheMazus, 2021-03-01:
You can just install the new version.

andreafurla, 2021-03-03:
Hi @TheMazus ,
yesterday evening I switched-off my MAC, and it rebooted by itself.
After this the system show me a box with many error codes.

Can I share them with you privately?

TheMazus, 2021-03-03:
Sure you can send me the error codes.

andreafurla, 2021-03-04:
Can you give me your email address?

closed 2021-04-20 by MatthiasKerbl