AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #156: Helper v10: not found!

xjzhong, 2021-03-02:
map 13 2020 4 thunderbolt
[image]

xjzhong, 2021-03-02:
Every time I open the app, I need to use the password to install helper tool, but it still doesn't work even though I install many times and restart the app/ the Mac many times.

TheMazus, 2021-03-03:
Hi @xjzhong,
have you tried the button "Reinstall Helper" as well?

xjzhong, 2021-03-04:
@TheMazus 
Yes, I tried many times, it still does not work.

TheMazus, 2021-03-04:
Which MacOS version do you have installed?

xjzhong, 2021-03-04:
@TheMazus 

Big Sur 11.2.1

It can use in the past. I have opened the function"Allow Discharge" and set the limitation to 80% before I sleep (the battery level is about 50% when I changed the settings), but I found the battery level is 93% in the next day. After that, even though I plugged in the power adaptor, the battery icon did not showed a charging state and the battery level kept decreasing. I  thought my Mac has any problems. I tried close AlDente, reinstall helper, reset SMC, and the Mac resumes normal use, but AlDente showed "Helper v10: not found" from then on.

TheMazus, 2021-03-07:
Hi @xjzhong,
try the following steps:
Quit and delete AlDente and go in Finder to Macintosh HD/Library/LaunchDaemons and delete “com.davidwernhart.Helper.plist”. Afterward, go to Macintosh HD/Library/PrivilegedHelperTools and delete “com.davidwernhart.Helper”. Restart your MacBook and install AlDente again.

xjzhong, 2021-03-09:
@TheMazus Sorry, it still doesn't work even though I do as you say.

jpirnie, 2021-03-11:
Try the latest version v2.1.2 - it fixed this problem for me.  It just came out today I think.

MrBoney, 2021-03-11:
I've just deleted all previous copies of AlDente, deleted the helper and helper .plist, rebooted and installed 2.1.2.
I now get the error 'Helper v12: not found!'

aniuwoo, 2021-03-11:
I got the same problem as MrBoney, and I did all the uninstallation-restart-reinstallation-restart thing, still missing Helper V12.

KyonCN, 2021-03-12:
I got the same problem too.
MacBook Pro 16 (2020) big sur 11.2.3
I encountered the problem when I updated the latest OS.
The app installed by brew is now tagged unsafe by apple, so I  have to run this`╰─ codesign --force --deep --sign - /Applications/AlDente.app` .

aniuwoo, 2021-03-18:
> I got the same problem too.
> MacBook Pro 16 (2020) big sur 11.2.3
> I encountered the problem when I updated the latest OS.
> The app installed with `homebrew` is now tagged unsafe by apple, so I have to run this`╰─ codesign --force --deep --sign - /Applications/AlDente.app` .
> 
> The problem is now solved by the latest release from Github. Thanks!

NO, the problem lingers with the latest version. Still missing Helper V12.

artinalbert, 2021-03-18:
I opened another issue and have exactly the same problem. Apple Silicon. Mac OS 11.2.3.
And also it can't find v12 helper and asks for my admin pass every time I reboot!

kiwi4814, 2021-03-26:
> I opened another issue and have exactly the same problem. Apple Silicon. Mac OS 11.2.3.
> And also it can't find v12 helper and asks for my admin pass every time I reboot!

I have the name problem!

MatthiasKerbl, 2021-07-05:
Hi everyone,

We just released a new version of AlDente which should solve the problem you encountered.
You can download the new version here: https://github.com/davidwernhart/AlDente/releases
Sorry, it took so long but I hope you like the new version.

Kind regards,

Matthias

closed 2021-07-05 by MatthiasKerbl