AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #157: Allow discharge: External monitor doesn't work when MacBook is closed

aktivde, 2021-03-02:
I just experienced an expectable bug: When having "allow discharge" enabled, an external monitor won't work when the Macbook is closed. Probably because the Macbook thinks not to be charged which generally prevents an external monitor to work if the Macbook is closed. 

No idea, if that's easy to fix or one needs to live with it. Just wanted to mention the issue.

vividfog, 2021-03-02:
#114 

Known issue. The download page mentions this, but the app itself doesn't. I also filed this once.

TheMazus, 2021-03-02:
Hi @aktivde,
as @vividfog already pointed out, this is a known problem and we are not sure if we can do anything about it.
We will definitely include a warning in the app in a future update.

closed 2021-03-02 by TheMazus