AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #158: MBA M1 can't display external display with lid closed / Idea for AlDente Pro

Jseeca, 2021-03-03:
I have a Dell U2720Q (Monitor) so I plug in a single USB C cable to hook up the display and all my peripherals. It works fine normally, just when you tick the "allow discharge" then there is no output at all once the charging percentage hits the specified limit. I know it is in beta, but just would like to point out the bug 

Also as a new feature idea for the upcoming pro AlDente. I think it would be a good idea to set variable limits. Batteries have an optimal storage percentage of around 40 percent. So that main idea is that the charging limit slowly decreases thought-out the day until your specified bed time at night. The charge limit then will be 40 percent and will remain that percentage for most of the night. Around an hour before you wake up the macbook will then automatically charge back to the normal limit again ready for you to use. This will further reduce battery degradation as the battery will always be one of two things. 

1) Discharging so it isn't in a high stressed state 
2) Being stored at the optimal battery percentage. 

I believe this would further improve the lifespan of the battery, as we are pretty much storing the battery at an optimal percentage for a large chunk of our day (8 hours of our day because of sleep) 

One more idea I think would be nice to add is to set boundaries for charging, this is called state of charge (SoC) This would be more applicable for people like me who use docks and keep there MB plugged in all the time. But pretty much you can also prolong your battery by charging your battery within specific percentages. Obviously if I capped it at 40 percent then it would be optimal, however I may need to take my laptop around with me and in that case 40 percent of charge is not enough so I need to keep the laptop at a higher percentage in the day. Therefore (according to battery university), if you charge between 65% - 75%, it will give you the most amount of cycles which I suggest you implement, rather than setting it to a fixed percentage. This would further improve it. 

Anyways I am very grateful that you have developed this app, I literally was holding off my M1 purchase until I knew there was M1 support for this app, as I know how bad charging your battery to 100 percent can be. I do not know why manufactures do no implement this natively to all electronic devices, its planned obsolesce. But thank you so much!!

TheMazus, 2021-03-03:
Hi @Jseeca,
the clamshell mode doesn't work when "Allow Discharge" is enabled (due to technical limitations).

Thank you very much for the feature ideas, we will definitely consider them.

I am happy to hear that you like AlDente. As you said, it is definitely planned obsolescence from manufacturers not to implement these features.

Best regards,
Matthias

closed 2021-03-03 by TheMazus

Jseeca, 2021-03-05:
Upon second thought, I just realised that variable limits won't be useful to me (but may to others) since, I use my laptop in clamshell mode most of the time... So I was thinking of another idea, would be to assign hotkeys for different limits. For example default my limit is set to 60%, but an hour before I leave the house, I can click on a hotkey to temporarily raise the limit to 80% or something.