AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #162: Charge limit not working in bootcamp

Swaggooo, 2021-03-05:
I'm on a 2019 MacBook Pro 16" with Al Dente v2.0 Alpha and the laptop charges to 100% in bootcamp even though I have my charge limit set to 70%.

Al Dente is working just fine under macOS.

TheMazus, 2021-03-05:
Hi @Swaggooo,
you have to switch to "Use Classic SMC Key (Intel)" for AlDente to work in Bootcamp.

Best regards,
Matthias

Swaggooo, 2021-03-05:
Hi Matthias,
The "Use Classic SMC Key (Intel)" switch has been enabled since I started using AlDente and it is still not working in Bootcamp.

If you have any other suggestions or if you need any other information in regards to this issue, please let me know.

Thanks

davidwernhart, 2021-03-05:
Hey @Swaggooo 

Please update to version 2.1.
On 2.0 Alpha, the Intel Key is reset everytime the app closes.
This has been fixed with version 2.1.

Hope that works for you
David

Swaggooo, 2021-03-05:
Hi David,

That did the trick for me! Thank you for the support :)

Swaggooo, 2021-03-23:
Hey guys,

Sorry to reopen this issue, but I seem to be facing this issue again. I am on Al Dente 2.1.2 and have tried reinstalling the helper as well.

My battery charge limit is set to 70%, and it functions as normal on macOS, but whenever I am on windows, it regularly seems to charge above that point. In the past week, I've noticed my MacBook is between 80-100% rather than staying below 70% as it should.

Thanks

MatthiasKerbl, 2021-03-23:
Hi @Swaggooo,
do you have "Use Classic SMC Key (Intel)" activated?

Swaggooo, 2021-03-23:
Hi Mattias,

Yes, I do

Swaggooo, 2021-04-02:
Hey all,

Just reporting back after using Al Dente v2.2 for about a week. The issue seems to no longer be present.

Thank you for all the support you provide! Looking forward to Al Dente Pro :)

closed 2021-04-02 by Swaggooo