AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #165: [Feature Request] Full-Charge scheduling and automation & long-work-day scheduling

Rastafabisch, 2021-03-06:
Coincidentally just some days before I came your awesome tool (a month ago via 9to5mac) I specifically searched for something providing this functionality and ended up empty. I really glad I eventually did come across.

What I thought would be a great addition is a function to schedule the once/twice in a month full recharge for maintenance (+ a notification to actually drain it, beforehand). 
Maybe this could even be auto-configured? Whenever the daemon recognises that the battery was drained below 15% (or so) it automatically triggers, that the next recharge should fill the battery, leave it there for two hours (or so) and then drop down to the charge level that is desired. This daemon could feature a history log, so that this full-charge-event would only be triggered twice a month (or so).

Additionally, though this might be better suited to be a follow up, the scheduling functionality could be expanded. If, for instance, Thursdays there is a higher demand (manually configured) to run longer on battery-power (starting from 7am) the daemon could be set to have the battery-level at a 100% at that time.  

I hope my ideas provide some helpful feedback and inspiration. Keep it up!

TheMazus, 2021-03-07:
Hi @Rastafabisch,

I am glad you like AlDente and thank you for these great feature ideas. We will definitely implement something in this regard.

Best regards,
Matthias

eighteentee, 2021-03-28:
This is a tremendous idea! I was just at the point of setting up a recurring calendar reminder to remind me to do a full charge once a month. Would definitely pay for this feature 👍

janandre, 2021-04-01:
+10

MatthiasKerbl, 2023-03-12:
Hi everyone,

We have just released AlDente 1.21.1 which has a new beta feature called Schedule. This feature allows users to set specific Actions in AlDente to run automatically on predetermined days, times, and intervals. For example, Calibration Mode can be set to run every two weeks.
You can read more about the Schedule feature and its current limitations here: https://apphousekitchen.com/feature-explanation-schedule/
You can find a full list of the Release Notes here: https://apphousekitchen.com/changelog/

Cheers,
Matthias

MatthiasKerbl, 2023-06-09:
Hi everyone,

Since our Schedule feature has officially been released for a while now, I will close this issue/feature request.

Best,
Matthias

closed 2023-06-09 by MatthiasKerbl