AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #170: After deinstallation: Battery stops charging and monitor is disconnected

aktivde, 2021-03-08:
I recently updated to MacOS 11.2.2 and got Aldente to work. However, after restarting it was always set to 100% again so I didn't see any sense in using it and deinstalled it.

Now I've still got the problem that the macbook – when it's closed and used with an external monitor – stops charging for a short time and loses connection to the external monitor. I know that because of the noise that appears when the charger connects again. It can also be proved by the battery management that tells the time the macbook is powered by the charger.

I used appcleaner to deinstall aldente so everything is removed. I also used apple's shortcuts to reset the SMC without any noticeable effect. 

Does anyone else have the same problem?

aktivde, 2021-03-09:
Update: The issue only appears when the charger is connected via the hub. When it's plugged in directly everything works fine. It rather seems to be an issue with the hub or with MacOS Big Sur 11.2.2.

closed 2021-03-09 by aktivde

TheMazus, 2021-03-09:
Hi @aktivde,

I'm sorry you have issues with AlDente. Normally, AlDente should remember the set Max. Battery Charge after a restart. Did you install AlDente in the applications folder? Do you have an Apple Silicon or Intel MacBook?

Best regards,
Matthias

aktivde, 2021-03-10:
Yes, Aldente was installed in the applications folder and I've got an Intel Macbook. I thought it was due to the update to 11.2.2...