AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #178: Al Dente is reported as malware by macOS 11.2.3

Averiun, 2021-03-10:
[image]

qfettes, 2021-03-10:
Same here. I can no longer open the app.

BigMcGuire, 2021-03-10:
On Macrumors we have some Silicon Mac users wondering how to reset the % now that they can't run AlDente anymore. No SMC reset like Intel Macs. Anyone know?

srbNYC, 2021-03-10:
I checked "disable malware protection," but it still says I do not have permission to open the app. Afraid my battery will be stuck at 63%!

hzozo, 2021-03-10:
Same here, I've just rebooted my MacBook this afternoon and faced the same thing. I've even overridden malware protection and removed the quarantine bit but that doesn't help.
I also share @BigMcGuire's concern, I'm just testing if my battery charges now to 100 or 73%.

Matt-136, 2021-03-10:
I was able to charge the battery to 100% but now it stays at 100 all the time

tomato990, 2021-03-10:
Same here. Big sur for macbook 2015.

Matt-136, 2021-03-10:
Coconut Battery is showing the right percentage (96%) but in the menu bar it stays at 100%

eltonooi, 2021-03-10:
Same here after restart my M1 Mac Air, Big Sur 11.2.1

davidwernhart, 2021-03-10:
Hey guys! Sorry for the inconveniences.
This problem seems to occur because my developer certificate has expired today.
I will update a freshly signed version soon.
David

srbNYC, 2021-03-10:
Relatively new to AlDente. Very impressed with your timely response and solve! Thank you!

TheMazus, 2021-03-10:
Hi guys,

The freshly signed version of AlDente was just released.
https://github.com/davidwernhart/AlDente/releases

Best regards,
Matthias

eltonooi, 2021-03-10:
> Hi guys,
> 
> The freshly signed version of AlDente was just released.
> https://github.com/davidwernhart/AlDente/releases
> 
> Best regards,
> Matthias

[image]
Hi, the apps can run now but this is the new warning

TheMazus, 2021-03-10:
> > Hi guys,
> > The freshly signed version of AlDente was just released.
> > https://github.com/davidwernhart/AlDente/releases
> > Best regards,
> > Matthias
> 
> [image]
> 
> Hi, the apps can run now but this is the new warning

Hi @eltonooi,
this is due to the old helper, which was signed with the expired developer certificate.
You just have to install the new version which should delete the old helpers and install the new ones. However, sometimes this won't work and you have to delete the old helpers manually as described in the release notes.

closed 2021-03-11 by Averiun