AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #181: Version 2.1 doesn't work on Catalina?

ngocphamm, 2021-03-10:
Downloading the version (https://github.com/davidwernhart/AlDente/releases/tag/v2.1) and I see that the icon is grayed out like this.
[image]

Trying to double click to open it results in this dialog box.
[image]

This is just a bit inconvenient, as I installed the app with `brew cask`, and now it's saying there's an update available for it. However, `brew upgrade` (that is supposed to install all available updates) doesn't know that the version is not compatible with my OS version.

TheMazus, 2021-03-10:
Hi @ngocphamm,

A new version of AlDente with Catalina support was just released.

Best regards,
Matthias

ngocphamm, 2021-03-10:
Thanks @TheMazus for the quick response to this.

I just tried the new release, and unfortunately it still doesn't work for me 😢

[image]

celsoazevedo, 2021-03-10:
Same here, v2.1.2 doesn't work on Catalina. I'm still using v1.2.

[image]

TheMazus, 2021-03-10:
Thanks for reporting back, we are looking into it.

davidwernhart, 2021-03-10:
Hey guys!
I think I made a small error building the current release.
Theoretically, the version attached to this post should run on Catalina.
Can you check if it works for you?
David

[AlDente_1.2.1_Catalina.app.zip](https://github.com/davidwernhart/AlDente/files/6119060/AlDente_1.2.1_Catalina.app.zip)

ngocphamm, 2021-03-10:
This app opens for me, and seems to be working! Thank you @davidwernhart

celsoazevedo, 2021-03-11:
> AlDente_1.2.1_Catalina.app.zip

This one works.

ngocphamm, 2021-03-11:
Update. I don't think the normal mode (new?) of AlDente 2.1.2 works for me (2019 16" MBP). My charging level was 59% with version 1.2. I set it to 70% with 2.1.2 and don't see it charging even after 10 minutes. I also tried restarting.

Once I set to "Use classic SMC Key (Intel)", it starts charging and now stays at 72%.

celsoazevedo, 2021-03-13:
Using the version David shared above, charging starts/stops with both options for me (also a 2019 MBP 16", running 10.15.7). The default method is more accurate than the "SMC Key" option (behaves like v1.2), but both work.

ngocphamm, 2021-03-13:
@celsoazevedo is right. It might be something else. I changed AlDente back to use the "normal" mode and it works now (setting at 40%, and it stopped charging just now at 41%).

celsoazevedo, 2021-03-15:
I don't know if this is a bug or just a limitation of the _new mode_, but the laptop charges to 100% if I shut it down and start charging. With the _Classic SMC Key_ mode, it stops normally.

I don't mind the inaccuracy of the classic mode, so this doesn't really affect me. Just sharing it here in case no one is aware.

closed 2021-04-05 by MatthiasKerbl