AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #183: macOS Big Sur - Reports AlDente as malware

FrancisVillarba, 2021-03-11:
macOS now reports AlDente as malware and prompts me to move the application to the bin, but I think this is a false positive based on the source code here. 

Is this related to the certificate issue? Seems a little dramatic is that's the case 😅

[image]

jpirnie, 2021-03-11:
Download and install the latest version v2.1.2 - it fixes this issue.  I just installed it and it's working great on a 2018 13" MBP.

CharlieBrown12, 2021-03-11:
Had this also - and learned it was about expired developer certificate.

**Any chance to know the date of the next certificate expiration?**

I keep my TimeMachine backups for a long time - and would be happy to know how frequently I need to update AlDente.

Many thanks!

patperek, 2021-03-11:
M1 MBP, Big Sur 11.2.3, v.2.1.2 works great.

davidwernhart, 2021-03-11:
@CharlieBrown12 
The current certificate is valid until 2026, although since this is now a paid Apple development certificate, I am pretty sure a renewal should be possible without the application to stop working.

Best regards,
David

closed 2021-03-11 by davidwernhart