AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #186: Blanks external monitor, WiFi unusable

dorkypants, 2021-03-11:
Just installed on 2015 12” Retina MacBook running 10.15.7. I normally run with external USB-C monitor and the MB cover closed. After launching AlDente, setting the limit to 80% and checking the “allow to drain” box, 2 things happened: 1) the external monitor blanked, 2) the WiFi went unstable; i.e., the signal strength was fine, but the menubar WiFi icon showed in its dropdown menu a constant fluctuation between “WiFi: On” and “WiFi: Looking for Networks…”. Even after following the instructions for uninstalling AlDente, WiFi is still hosed. Resetting SMC Controller and NVRAM made no difference. Booting in Safe Mode made no difference

dorkypants, 2021-03-12:
> Hi @dorkypants,
>
> the feature "Allow Discharge" does not support the clamshell mode. This is described in our README, and I have updated our release notes as well so people are more likely to see it. However, no one else who tried "Allow Discharge" experienced problems with their WiFi. Do you still have issues with your WiFi?
>
> Best regards,
> Matthias

Matthias,

Unfortunately the WiFi remains unusable despite assistance via online chat from Apple Support. I have a Genius Bar appointment coming up. They may be able to resolve it.

Michael 

အသီးမှာသရက်၊ အသားမှာဝက်၊ အရွက်မှာလက်ဖက်။

TheMazus, 2021-03-12:
> Matthias, Unfortunately the WiFi remains unusable despite assistance via online chat from Apple Support. I have a Genius Bar appointment coming up. They may be able to resolve it. Michael အသီးမှာသရက်၊ အသားမှာဝက်၊ အရွက်မှာလက်ဖက်။
> […](#)

Thank you for the update and please report back after your Genius Bar visit.

MatthiasKerbl, 2021-04-20:
Hi Michael,

since you haven't reported back, I guess the issue was not related to AlDente. Therefore, I will close this issue for now.

closed 2021-04-20 by MatthiasKerbl

dorkypants, 2021-04-20:
Hi Martin,

I haven’t been able to use my MacBook since the reported problem and am now on an M1 Mac Mini, which of course has no need for AlDente. 

After consulting with Apple Support via online chat about the MacBook not being able to connect to WiFi (even in Safe Mode), they recommended taking to an Apple Store for further diagnosis. I made a Time Machine backup in case they had to reinstall macOS. It turned out they had to wipe the machine clean and install macOS.

I took the opportunity to have a replacement battery installed by an independent shop. There were complications, such that more than a month later, I still do not have it back. That’s why I moved to an M1 Mac Mini. 

The curious thing is that after restoring my Time Machine backup to the Mini, neither its WiFi nor wired Ethernet worked, except in Safe Mode. Adding back my Login Items didn’t disrupt WiFi. It appears that a LaunchAgent or LaunchDaemon installed by a 3rd party utility may be the culprit: LittleSnitch or WhatRoute. Since I only tried them briefly and don’t use them, I’ve disabled these. Is it possible that, on my MacBook, installing AlDente triggered an adverse interaction with one of them?

Michael