AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #187: "AlDente" will damage your computer. You should move it to Trash.

Maxonyks, 2021-03-12:
[image]

Alright, I restarted my computer and this shows up. I hit cancel and after a few seconds, I came up right back.

Maxonyks, 2021-03-12:
Okay I just downloaded the new update and it fixes the problem, but still there is another warning sign that pops up, is this safe? Thank you in advance.
[image]

AdrianLicu, 2021-03-12:
The version from homebrew still gives the first dialog. Had to download the one from github to allow me run.

Regards

TheMazus, 2021-03-12:
Hi @Maxonyks,
here is the clarification on this issue: Due to an expired developer certificate, all versions of AlDente prior to 2.1.2 are not working anymore. We have just released an updated and fixed version (AlDente 2.1.2, https://github.com/davidwernhart/AlDente/releases) signed with our new official developer account. However, some might need to delete the old helper files manually (instructions are in the release notes).

TheMazus, 2021-03-12:
> The version from homebrew still gives the first dialog. Had to download the one from github to allow me run.
> 
> Regards

Hi @AdrianLicu,
since homebrew has not updated to our new version, it includes our old version which is signed with our expired developer account. Therefore, it can't be used anymore.

closed 2021-03-12 by TheMazus