AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #188: Doesn’t work on Catalina 10.15.7 anymore

23ZZY3, 2021-03-12:
I appreciate your work and it’s a good application.
First I’m using the 1.2 version and it work perfectly on my 2020 MacBook Pro with 4 ports(intel). But i just find it doesn’t work today. And i tried your suggestions (update to new version and delete plist...). Whatever i set the battery won’t charge. Then i find there’s a supplement update that solving the 2016/2017 model charging issues. After i installed this update, your application seems to be not exist. It will charge to 100 per like the app was never installed. My solution now is using sudo ./smc -k BCLM -w <0x> In terminal in this manual way.
Anyway thanks for your project.

TheMazus, 2021-03-14:
Hi @23ZZY3,
I am sorry you have trouble with our new version. Other people reported it is working fine on Catalina. Have you tried the "Use Classic SMC Key (Intel)" mode?

Matthias

RYANTANGCN, 2021-03-27:
I have the same issue on my macbook pro (13-inch, 2019, Two Thunderbolt 3 ports), catalina version is 10.15.7;No matter check or uncheck "use classic SMC key(intel)" ,it still "helper v12 not found". I follow these steps #

RYANTANGCN, 2021-03-27:
I have the same issue on my macbook pro (13-inch, 2019, Two Thunderbolt 3 ports), Catalina version is **10.15.7**. I follow these [steps](https://github.com/davidwernhart/AlDente/issues/204#issuecomment-802311643) and reinstall helper many times, no matter check or uncheck "use classic SMC key(intel)", it still not working.

MatthiasKerbl, 2021-07-05:
Hi everyone,

We just released a new version of AlDente which should solve the problem you encountered.
Unfortunately, the new version is only compatible with macOS 11 Big Sur.
You can download the new version here: https://github.com/davidwernhart/AlDente/releases

Kind regards,
Matthias

closed 2021-07-05 by MatthiasKerbl