AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #192: Consider removing the popover animation?

sakesalverda, 2021-03-13:
Hi, I noticed the AlDente popover uses an animation when opened. Perhaps it can be disabled to bring it more in line with other apps/the system. I believe this can be easily done by setting the popover animates property
And thanks for the project, really enjoying the app!!

TheMazus, 2021-03-15:
Hi @sakesalverda,
I am glad you like our app. Maybe we will give the functionality to deactivate the animation.

Matthias

MatthiasKerbl, 2021-07-05:
Hi @sakesalverda,

We just released a new version of AlDente in which we disabled the popover animation.
You can download the new version here: https://github.com/davidwernhart/AlDente/releases

Kind regards,
Matthias

closed 2021-07-05 by MatthiasKerbl

sakesalverda, 2021-07-05:
Hi @MatthiasKerbl thank you for keeping me up to date, appreciate that! And again, thank you for the awesome app! Still using it everyday

MatthiasKerbl, 2021-07-05:
> Hi @MatthiasKerbl thank you for keeping me up to date, appreciate that! And again, thank you for the awesome app! Still using it everyday

I am glad you still like it. If you have more ideas let us know and if you have not heard about our Pro version yet, you can check it out on our website: https://apphousekitchen.com/