AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #201: Question. Should I disable "Optimised battery Charging" in settings?

porcherie, 2021-03-15:
Are you aware of any possible conflict it both al dente and Optimised battery Charging are enable?

I am on Big Sur 11.2.3 and MacBook Pro m1. So far working great. I just turn it on, set the amount and off again to avoid the app to use CPU (it was around 3-4% for me)

Brewforce, 2021-03-17:
Hello, im on M1 MBA with Big Sur 11.2.3, and mine it's using only 0,2 % of CPU with the last release. Just to add some more data to the thread. I'm also concerned about the coexistence of both al dente and the Optimised Battery Charging option.

MatthiasKerbl, 2021-03-17:
Hi @porcherie,
some people reported conflicts with "Optimised battery Charging" activated and AlDente. Therefore, we recommend disabling it.

Matthias

closed 2021-03-17 by MatthiasKerbl

porcherie, 2021-03-17:
> Hi @porcherie,
> some people reported conflicts with "Optimised battery Charging" activated and AlDente. Therefore, we recommend disabling it.
> 
> Matthias

Thanks for your answer. I will turn it off then