AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #203: waiting for pro :-))

tosbsas, 2021-03-18:
Doesnt't happen often but I am ready to spend some money on such a great programm - and Pro sounds awesome.

Any news on this, on what's gonna be Pro and stuff?

MatthiasKerbl, 2021-03-18:
Hi @tosbsas,
we are currently developing AlDente Pro and making great progress. However, it still takes time until it is finished.
Do you have any feature ideas you would like to see?

Matthias

tosbsas, 2021-03-19:
:-)) hmm thats a tough question for someone with no experience in whats possible. I would like a trustworthy battery health indicator with % of design capacity, a sensor for the battery temperature and :-))

An option to set the max value to 100 without having the move the slider - so it would charge my MBP to 100% before an event where I will use the MBP on battery. Like a - disable the 80% you have set for this occasion and let battery charge to 100%. Now I have to open aldente, move the slider and remind myself that I need to move it again. Dunno if you get my idea

MatthiasKerbl, 2021-03-20:
Unfortunately, the battery health indicator is never really trustworthy. You get the most accurate readout when you cycle your battery from 0%-100%-0% a couple of times. However, normally we would like to avoid this behavior to avoid unnecessary cycles.
The other features might be possible and I will discuss them with David about implementing them. :)

tosbsas, 2021-03-20:
Ok. I understand. You are saying really go down to 0. How often? Once a week?

Would be awesome to the see the dísable one Feature :-)

tosbsas, 2021-03-20:
Aber ihr macht einen Hammer Job. You’ll get my money without Any doubt

MatthiasKerbl, 2021-03-20:
Going down to 0% is the best for recalibrating but the worst for your battery. I would do a full cycle at least once a month.
Vielen Dank für die Unterstützung. :)

tosbsas, 2021-03-21:
👍🏼

MatthiasKerbl, 2021-04-16:
Hi @tosbsas,

since you got the pro version already, I think we can close this issue. :)

closed 2021-04-16 by MatthiasKerbl