AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #204: Unfortunately doesn't work on MBP

patryk-jason, 2021-03-18:
Since MacOS showed up a warning (AlDente will damage your computer. You should move it to the trash) Aldente doesn't work anymore. Using new version 2.1.2 on MBP M1 2020. Battery keep charging when limit is set up.

elibooklover, 2021-03-18:
I am having the same issue on my MBA M1.

MatthiasKerbl, 2021-03-18:
Hi,
please try the following steps:

1. Set Max. Charge to 100% and wait 30 seconds.
2. Quit AlDente and delete the program file.
3. Go in Finder to Macintosh HD/Library/LaunchDaemons and delete “com.davidwernhart.Helper.plist”.
4. Go to Macintosh HD/Library/PrivilegedHelperTools and delete “com.davidwernhart.Helper”.
5. Restart your MacBook.
6. Download and install AlDente again.
7. Restart your MacBook again.

Hope this helps,
Matthias

longalong, 2021-03-21:
> Hi,
> please try the following steps:
>
> 1. Set Max. Charge to 100% and wait 30 seconds.
> 2. Quit AlDente and delete the program file.
> 3. Go in Finder to Macintosh HD/Library/LaunchDaemons and delete [...]

I am having the same issue on my MBA M1.  I have tried these steps for many times, it always show “helper V12 not found” since reboot my MBA. it didnot work on bigsur 11.2.3

patryk-jason, 2021-03-22:
> Hi,
> please try the following steps:
>
> 1. Set Max. Charge to 100% and wait 30 seconds.
> 2. Quit AlDente and delete the program file.
> 3. Go in Finder to Macintosh HD/Library/LaunchDaemons and delete [...]

I went through all steps. I didn't have any files in PrivilegedHelperTools folder. 
It is working now 🙂

MatthiasKerbl, 2021-03-22:
> I went through all steps. I didn't have any files in PrivilegedHelperTools folder.
> It is working now 🙂

I am glad it helped and thanks for the feedback.

longalong, 2021-03-31:
with 2.2 version。I am having the same issue on my MBA M1. I have tried these steps for many times, it always show “helper V12 not found” since reboot my MBA. it didnot work on bigsur 11.2.3

FT180222, 2021-05-04:
sudo launchctl load -w /Library/LaunchDaemons/com.davidwernhart.Helper.plist 终端输入这个完美解决。
[image]

MatthiasKerbl, 2021-05-04:
> sudo launchctl load -w /Library/LaunchDaemons/com.davidwernhart.Helper.plist 终端输入这个完美解决。
> [image]

Hi @FT180222,
thanks for this suggestion.

MatthiasKerbl, 2021-07-05:
Hi @patryk-jason ,

We just released a new version of AlDente which should solve the problem you encountered.
You can download the new version here: https://github.com/davidwernhart/AlDente/releases
Sorry, it took so long but I hope you like the new version.

Kind regards,

Matthias

closed 2021-07-05 by MatthiasKerbl