AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #205: Asks for Admin Password  to install the helper program after each RESTART!! Can't install it.

artinalbert, 2021-03-18:
11.2.3 Big Sur
2.1.2 AlDente

MatthiasKerbl, 2021-03-18:
Hi @artinalbert,
please try the following steps:

1. Quit AlDente and delete the program file.
2. Go in Finder to Macintosh HD/Library/LaunchDaemons and delete “com.davidwernhart.Helper.plist”. 
3. Go to Macintosh HD/Library/PrivilegedHelperTools and delete “com.davidwernhart.Helper”.
4. Restart your MacBook.
5. Download and install AlDente again.
6. Restart your MacBook again.

Hope this helps,
Matthias

aniuwoo, 2021-03-19:
> Hi @artinalbert,
> please try the following steps:
>
>     1. Quit AlDente and delete the program file.
>
>     2. Go in Finder to Macintosh HD/Library/LaunchDaemons and delete [...]

Follow the steps every time for several times, the problem is still there. No Helper V.12 found. 11.2.2 Big Sur, M1 MBP

artinalbert, 2021-03-20:
I followed suit as well. Same issue persists.  

> Hi @artinalbert,
> please try the following steps:
>
> 1. Quit AlDente and delete the program file.
> 2. Go in Finder to Macintosh HD/Library/LaunchDaemons and delete “com.davidwernhart.Helper.plist”.
> 3. [...]

longalong, 2021-03-21:
I followed suit as well. Same issue persists.，No Helper V.12 found. 11.2.2 Big Sur, M1 MBP

FT180222, 2021-05-04:
sudo launchctl load -w /Library/LaunchDaemons/com.davidwernhart.Helper.plist 终端输入这个完美解决。
[image]

martindon6, 2021-05-04:
I has the 2019 macbook pro intel version, I tried your method but I still cannot instal the helper V12.

[image]

FT180222, 2021-05-04:
你试试把第二行打勾，我的是m1。

eflylab, 2021-05-14:
sudo launchctl load -w /Library/LaunchDaemons/com.davidwernhart.Helper.plist

FT180222, 2021-05-14:
电脑上的终端输入 命令

MatthiasKerbl, 2021-07-05:
Hi everyone,

We just released a new version of AlDente which should solve the problems you encountered.
You can download the new version here: https://github.com/davidwernhart/AlDente/releases
Sorry, it took so long but I hope you like the new version.

Kind regards,

Matthias

closed 2021-07-05 by MatthiasKerbl