AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #207: Battery status will stay 100% usage for a long time, and then run out of power gradually at 64%

halqin, 2021-03-19:
About two weeks ago, I notice if I use battery, my MacBook battery status will stay 100% usage for a long time, and then run out of power at 64% gradually (shut down without any battery warning). I have to connect adapter to wake up computer. 

The effort have tried so far: 
-  reset SMC many times
- Set AlDente to maximum and then uninstall 
- Update to 2.1.2 and then redo above trials 
- Reinstall macOS  and reset SMC again 

Still not working. 
I use MacBook Pro 2017, 
BigSur 11.2.3.
64% is the threshold I used to use.

MatthiasKerbl, 2021-03-19:
Hi @halqin,
There are two reasons why your battery behaves like this. 1. Your battery calibration is disturbed (this is reversible) or 2. your battery has degraded because of usage and time (this is not reversible).
Please answer the following questions and we might find out what the cause is.

1. Since when do you have the MacBook?
2. Since when do you use AlDente?
3. What was your battery usage before using AlDente? Did you use your MacBook while it was plugged in most of the time or did you use the MacBook on "battery mode"?
4. What was your battery usage with AlDente? Did you let it plugged in most of the time or did you use it in "battery mode" as well once a week?
5. When was the last time you completely cycled your battery from 0% to 100%?
6. What does your MacOS battery status show?
7. How many charging cycles does your battery have?
8. What's your remaining Full Charge Capacity?

westrayhao, 2021-03-19:
Hi @MatthiasKerbl ,

Thanks for your quick response! 

1. I use the macbook pro 2017 since 2017 October. 
2. I use ALDente less than 2 months  
3. I often use power adapter before AlDente
4. I often use poser adapter with AlDente
5. Maybe two month ago. At least once since this year.
6. Battery Information:
  Model Information:
  Manufacturer:	DSY
  Device Name:	bq20z451
  Pack Lot Code:	0
  PCB Lot Code:	0
  Firmware Version:	901
  Hardware Revision:	1
  Cell Revision:	3666
  Charge Information:
  Fully Charged:	Yes
  Charging:	No
  Full Charge Capacity (mAh):	4022
  State of Charge (%):	100
  Health Information:
  Cycle Count:	295
  Condition:	Normal

I fixed my keyboard by Mac in 2019, so my battery was also replaced with a new one at that time. I try to do  battery calibration, the laptop will turn off at 64%, and it is also not working after I charge long enough time. The battery condition is ok, I could still have about 4-5 hour duration, but the measurement in OS is not correct anymore. It starts charging from 64%

halqin, 2021-03-19:
More information about the battery:
Health: 85%

leeummm, 2021-03-19:
It sounds like your battery is fine. I would try and cycle it down to 20-30% a few times, reboot, and maybe reset SMC.

halqin, 2021-03-19:
I have done it several times, I use it until shut down and then charge it more than 3 hours. And reset SMC more than 10 times in total. Still not working.

MatthiasKerbl, 2021-03-19:
> Hi @MatthiasKerbl ,
> 
> Thanks for your quick response!
> 
> I use the macbook pro 2017 since 2017 October.
> I use ALDente less than 2 months
> I often use power adapter before AlDente
> I often use poser adapter with AlDente
> Maybe two month ago. At least once since this year.
> Battery Information:
> Model Information:
> Manufacturer: DSY
> Device Name: bq20z451
> Pack Lot Code: 0
> PCB Lot Code: 0
> Firmware Version: 901
> Hardware Revision: 1
> Cell Revision: 3666
> Charge Information:
> Fully Charged: Yes
> Charging: No
> Full Charge Capacity (mAh): 4022
> State of Charge (%): 100
> Health Information:
> Cycle Count: 295
> Condition: Normal
> I fixed my keyboard officially in 2019, so my battery was also replaced with a new one at that time. I try to do battery calibration, the laptop will turn off at 64%, and it is also not working after I charge long enough time. The battery condition is ok, I could still have about 4 hours duration, but the measurement in OS is not correct anymore. It starts charging from 64%

I think it is just a disturbed battery calibration since cycling your battery only once in the last three months is just not enough. I know you tried to cycle your battery already a couple of times, however, I would recommend cycling it even more often to recalibrate your BMS.

halqin, 2021-03-20:
Hey, Thanks! 

> I think it is just a disturbed battery calibration since cycling your battery only once in the last three months is just not enough. I know you tried to cycle your battery already a couple of times, [...]

Hey, Thanks! The battery back to normal today after I do calibration several times (3-4 times). I realise  that I ignore the "IMPORTANT" in your readme. And once more thing want to confirm, does calibration per month could avoid this problem?

closed 2021-03-20 by halqin

MatthiasKerbl, 2021-03-20:
Hi, I am glad your battery went back to normal. Yes, cycling your battery from 100%-0%-100% once or twice a month will avoid this issue.

vassudanagunta, 2021-03-21:
@MatthiasKerbl is it necessary to  go all the way to 0%? Would going to 20% or even 10% be enough?  I've read before that is bad for the battery, but maybe that's outdated info?

halqin, 2021-03-21:
If you often use adapter, the answer is yes. But of course you should not do it too often, otherwise it will damage your Lion battery. 
https://help.ifixit.com/article/265-battery-calibration

vassudanagunta, 2021-03-21:
Ok, thanks.  I read the part about why its necessary to go to total drain and shutdown.  

Second question, is it necessary to do this once a month (or more)?  I can I go months 4 months without the 100% to 0% recalibration, then do it and be no worse off than if I did it monthly?   Especially id I didn't care if I didn't have the full battery range during those 4 months?

halqin, 2021-03-21:
I got the problem without doing calibration 3 months. If you use this software and you often use adapter, you may have this problem. But you can try 4 month if you want. If something wrong it is still reversible anyway. All these discussion according to my experience, just a reference.

vassudanagunta, 2021-03-21:
Well that's the question. Is it 100% reversible, or is there some irreversible loss if I wait 4 months to recalibrate?  Maybe no one really knows the answer in any truly scientific way... Are we all, including the makers of AlDente, working off of anecdotal data, especially when it comes to recommended charge ranges and recalibration frequency?

Stone-JI, 2022-02-02:
> Hey, Thanks!
>
>
> Hey, Thanks! The battery back to normal today after I do calibration several times (3-4 times). I realise that I ignore the "IMPORTANT" in your readme. And once more thing want to [...]

Hey Halqin,
Same situation happens to my macbook air m1 also, mine stays at 100 and dies at 40, could you please explain how exactly you fixed your mac, it would be so helpful for me, thanks!

MatthiasKerbl, 2022-02-03:
> Hey Halqin, Same situation happens to my macbook air m1 also, mine stays at 100 and dies at 40, could you please explain how exactly you fixed your mac, it would be so helpful for me, thanks!

Hi @Stone-JI,

We have an explanation for this problem in our FAQ:

HELP MY BATTERY CAPACITY DROPPED AND MY MACBOOK TURNS OFF WITH 50% LEFT
Keeping your battery at a lower percentage, such as 20-70%, over weeks without doing full cycles (100%-0%-100%) can result in a disturbed battery calibration. When this happens, your Macbook might turn off with 40-50% left or your battery capacity will drop significantly. However, this is only due to a disturbed battery calibration and not because of a faulty or degraded battery. To avoid this issue, we recommend doing at least one full cycle (0%-100%) every two weeks. Even if your battery calibration gets disturbed, doing 4-5 full cycles will recalibrate your battery and the capacity will go up again. AlDente Pro has a feature called Calibration Mode(https://apphousekitchen.com/feature-explanation-calibration-mode-2/), which will automatically do a full cycle when startet. You can read more about the calibration of "smart" batteries here: https://batteryuniversity.com/article/bu-603-how-to-calibrate-a-smart-battery

Cheers,
Matthias