AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #214: M1 Macbook Pro shutting off at ~30%

RaymondDashWu, 2021-03-21:
I suspect that the latest version of Aldente (2.1.2) is causing my MBP to shut off prematurely. For the past 2 days since I've downloaded it it's shut down when it reaches ~30%. It will boot loop and won't turn back on until I connect it to a power source. After I do it'll read that same percent as before it shut down (34% as in today's case).

To confirm this I'm trying to figure out where the helper is located. There doesn't seem to be any files matching "davidwernhart" in my Library folder anymore so I'm guessing this has changed. Where is the new helper located?

MatthiasKerbl, 2021-03-21:
Hi @RaymondDashWu,
It sounds like your battery calibration is disturbed. Take a look at this issue: #207

The Helpers are located here:
/Library/LaunchDaemons/com.davidwernhart.Helper.plist
/Library/PrivilegedHelperTools/com.davidwernhart.Helper

RaymondDashWu, 2021-03-21:
Ah I hadn't seen that. Also mentions that in the FAQ. I'll close this for now and if I don't make another issue or reopen this in the future that was the solution.

closed 2021-03-21 by RaymondDashWu