AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #215: Aldente helper kept crash in M1 Macbook

xcity, 2021-03-21:
I am being experienced this for several times, normally, it happens during reboot, and cause the reboot failure. next reboot might solve this problem. the console logs shows below info. 

Process:               com.davidwernhart.Helper [81957]
Path:                  /Library/PrivilegedHelperTools/com.davidwernhart.Helper
Identifier:            com.davidwernhart.Helper
Version:               12
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
Responsible:           com.davidwernhart.Helper [81957]
User ID:               0

Date/Time:             2021-03-20 23:11:22.971 +0800
OS Version:            macOS 11.2.3 (20D91)
Report Version:        12
Anonymous UUID:        9C6CAD96-DA46-F28A-5ABF-D4C054236EAB

Sleep/Wake UUID:       33052AC3-B838-4C2A-958A-64BE77D5142C

Time Awake Since Boot: 150000 seconds
Time Since Wake:       7500 seconds

System Integrity Protection: enabled

Crashed Thread:        1  Dispatch queue: com.apple.NSXPCConnection.user.com.davidwernhart.Helper.mach.708

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       EXC_ARM_BREAKPOINT at 0x00000001004d52ac (brk 1)
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Trace/BPT trap: 5
Termination Reason:    Namespace SIGNAL, Code 0x5
Terminating Process:   exc handler [81957]

MatthiasKerbl, 2021-03-26:
Hi @xcity,
do you get the same issue with our newest version AlDente 2.2?

prwnr, 2021-04-06:
@MatthiasKerbl Im not the original reporter here, but I can say that I am getting that crash on v2.2 too. 

```Process:               com.davidwernhart.Helper [4539]
Path:                  /Library/PrivilegedHelperTools/com.davidwernhart.Helper
Identifier:            com.davidwernhart.Helper
Version:               12
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
Responsible:           com.davidwernhart.Helper [4539]
User ID:               0

Date/Time:             2021-04-05 20:12:40.085 +0200
OS Version:            macOS 11.2.2 (20D80)
Report Version:        12
Anonymous UUID:        0C416C47-0C23-3BC0-19E9-92E905A597C2

Sleep/Wake UUID:       CE42013E-FEA6-4BCC-AF31-4D71669FC39E

Time Awake Since Boot: 7300 seconds
Time Since Wake:       390 seconds

System Integrity Protection: enabled

Crashed Thread:        1  Dispatch queue: com.apple.NSXPCConnection.user.com.davidwernhart.Helper.mach.3260

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       EXC_ARM_BREAKPOINT at 0x0000000104a9129c (brk 1)
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Trace/BPT trap: 5
Termination Reason:    Namespace SIGNAL, Code 0x5
Terminating Process:   exc handler [4539]
```

xcity, 2021-04-13:
> Hi @xcity,
> do you get the same issue with our newest version AlDente 2.2?

Still reported with 2.2, and last MacOS.

Roman2K, 2021-04-13:
Same, still reported with v2.2.
6 crashes over the last 2 days: https://gist.github.com/Roman2K/237c7242c6a376104784fbb99ce01c76

MatthiasKerbl, 2021-04-13:
Hi,
Thanks for reporting this bug. We are looking into this issue and I will release an update when we fix it.

MatthiasKerbl, 2021-07-05:
Hi everyone,

We just released a new version of AlDente which should solve the problem you encountered.
You can download the new version here: https://github.com/davidwernhart/AlDente/releases
Sorry, it took so long but I hope you like the new version.

Kind regards,

Matthias

closed 2021-07-05 by MatthiasKerbl