AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #218: Allow discharge --> connection to Monitor is interrupted.

eliah22, 2021-03-22:
Hello first of all thanks for this wonderful tool. To my issue I connected my Macbook pro m1 to an LG monitor. When I click on allow discharge, the connection to the mac book pro m1 and the Monitor is disabled and it says No connection.  I use a usb c cable to connect.

patryk-jason, 2021-03-22:
> Experimental option to automatically discharge to desired percentage when plugged in. (Does not work in clamshell mode!)

Is your Macbook opened or closed while it happens?

eliah22, 2021-03-22:
Hey Patryk-Jason, thank you for the reply my Mac book is closed.

MatthiasKerbl, 2021-03-22:
Hi @patryk-jason,
Allow Discharge is not supported while your MacBook is in clamshell mode. Hence, it's only a "Beta" feature. However, you can just deactivate Allow Discharge and use Aldente even in clamshell mode.

Matthias

closed 2021-03-22 by eliah22