AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #219: Feature Request: Provide an Option for a Start-Charging Ratio

yanzhen0923, 2021-03-24:
Hi,

What I observe from the app is, if I set the max charge ratio to 64% for example, the app would then start charging when the capacity is 60 to 63%, I'm wondering now,

1. Can we get a new feature, this allows the user to set something like a "Min charge ratio", say 40%, then it would still stop charing when the capacity reaches 64%, but would not start charging until the capacity goes below 40%, so that the battery would go into a 40% ~ 64% cycle， not a 63% ~ 64% cycle

2. Is the 40% ~ 64% cycle，better than the 63% ~ 64% cycle for the battery health?

Best

tosbsas, 2021-03-25:
love the idea

danfoerster, 2021-03-26:
Til then, you can just set the max charge level to 64, wait til it stops charging at that percentage and then set the value in AlDente to the lower value

yanzhen0923, 2021-03-26:
> Til then, you can just set the max charge level to 64, wait til it stops charging at that percentage and then set the value in AlDente to the lower value

Yes sure, but this not quite automatic ;)

danfoerster, 2021-03-26:
Of course not, that's why I said 'til then' (hopefully the Pro version or future updates will have all those features). It's the closest thing you can do right now. I have my Mac plugged in pretty much 24/7 and the battery has not been charged since a few weeks. So the method does work quite well. To set the lower value is just so the top-off charges are basically disabled. The thing with AlDente is you have to experiment a lot to find out what works best for you, what is practical and somewhat reasonably healthy for the battery

MatthiasKerbl, 2021-07-05:
Hi @yanzhen0923,

we introduced this feature with AlDente Pro and it is called Sailing Mode.
You can check it out here: https://apphousekitchen.com/

Thank you for this idea!
Matthias

closed 2021-07-05 by MatthiasKerbl