AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #225: High CPU usage / memory leak?

aleeraser, 2021-03-28:
In the last 4-5 days I left my macbook constantly on for some time-intensive tests for a project (i.e. no idle/clamshell sleep).

I noticed already 3 times that after a while (can't tell exactly when) Aldente UI starts acting very slow and the CPU usage is high, too much much for a charge limiter application. All the times I had to quit and restart the app.

Here's a screenshot of Activity Monitor.

[image]

Here are my settings:

[image]

p.s. if the app is closed, does the limit keep working?

MatthiasKerbl, 2021-03-28:
Hi @aleeraser,

this was a known bug of the older AlDente versions. Just install our newest version and this issue will be gone.

aleeraser, 2021-03-28:
Thanks @MatthiasKerbl,
I didn't know there was a newer version!
Nice that the new version also has a check for updates.

Thanks

closed 2021-03-28 by aleeraser