AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #226: Using "able to discharge" behavior

tosbsas, 2021-03-28:
Hi

I understood than when using allow discharge, if I turn the MBP completely off, battery will charge to 100% anyway.

But
During the day I get several "connect" "disconnect" sounds, and battery goes below 80% to be recharged to 80% (my threshold). Is that normal?

danfoerster, 2021-03-29:
The allow discharge feature does one thing only. If your Mac is connected to power and the battery is higher than the value you set in AlDente, it will discharge to this value. I don't know how it works if the Mac is turned off. However if your Mac is running and you have the Optimized Battery Charging setting enabled in macOS settings, the two things might conflict with each other, meaning one might want to charge and the other discharges as soon as one recent is gained. I think you might have to turn of the Battery Managing feature in macOS Battery settings. Give it a try

tosbsas, 2021-03-29:
The optimize battery feature is disabled. My guess is that the allow to discharge feature checks once in a while if the percentage threshold is reached and turns power on or off. But its happening more often here than on 2.1.2

MatthiasKerbl, 2021-03-29:
> Hi
> 
> I understood than when using allow discharge, if I turn the MBP completely off, battery will charge to 100% anyway.
> 
> But
> During the day I get several "connect" "disconnect" sounds, and battery goes below 80% to be recharged to 80% (my threshold). Is that normal?

Hi @tosbsas,
this is currently the "normal" behavior of Allow Discharge. However, we are working on improving this in an future update.

tosbsas, 2021-03-29:
ok, you referring to both of my observations, right?

MatthiasKerbl, 2021-03-29:
> ok, you referring to both of my observations, right?

Yes

closed 2021-03-29 by tosbsas