AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #228: 13-inch 2020 4x TB ports Big Sur 11.2.3 on USB-C dock charging to full after a while

markouustalu, 2021-04-01:
It works on my other older laptop flawlessly. But not in my newer laptop. I install it and set it to 60%. At home it manages to keep it nicely on 60%. I think closing the lid and letting it go to standby ruins it after a while. But it is manageable.

At my work i have USB-C Dock. I set it to 60%, it shows that it is working on battery. Music is playing all the time so no standby issues. After a while it is charging again and battery is full as shown on the image.
[image]

Tried restarts and re-installs, nothing seems to help.

How can i help to solve this?

bruno-briner, 2021-04-05:
It was also happening with my MacBook Pro 13" late 2017 4 TB ports. I could only make it work reliably by selecting "Use Classic SMC Key (Intel)".

MatthiasKerbl, 2021-04-05:
> It works on my other older laptop flawlessly. But not in my newer laptop. I install it and set it to 60%. At home it manages to keep it nicely on 60%. I think closing the lid and letting it go to [...]

Hi @markouustalu,
have you tried it with "Allow Discharge" Disabled?

markouustalu, 2021-04-06:
Removed Allow Discharge. Still charging.
[image]
Tried Classic SMC key. Still Charging.
[image]

markouustalu, 2021-04-06:
after restarting the app i left the checkpoints checked as in my first post. It discharged my battery to about 90% and now it is charging again, no stand by's or lid closings in the mean time.

markouustalu, 2021-04-09:
guys, after multiple app and mac restarts and using the Classic SMC key it seems to have stopped charging at the set level. It seems pretty fragile but it seems to be working. If there are no more questions then this issue can be closed. If still needed, i would be more than happy to provide some debugging info to make this cool app as solid as a rock.

MatthiasKerbl, 2021-07-05:
Hi @markouustalu,

We just released a new version of AlDente which should make AlDente much more relyable.
You can download the new version here: https://github.com/davidwernhart/AlDente/releases

Kind regards,
Matthias

closed 2021-07-05 by MatthiasKerbl