AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #229: Max. charge often resets to 100% after sleep/shutdown on Macbook Pro 13" 2020 Intel

ernst42, 2021-04-03:
I have a Macbook Pro 13" 2020 Intel with touch bar and 4 TB ports, 2,3 GHz Quad-Core Intel Core i7, 32 GB RAM running Catalina 10.15.7 (19H524).

Now I am using AlDente 2.2, before I was using 1.2, max. charge set to 70%,  both versions showed the same problem:

After sleep for a few hours / wakeup and also after shutdown / restart the max. charge percentage gets reset back to 100% making AlDente useless because the battery will charge to full capacity in that case.

I have "Use Classic SMC key" checked, because of https://github.com/davidwernhart/AlDente/issues/209

ernst42, 2021-04-04:
I think during sleep the problem only occurs if the the RAM is powered down (real hibernation) and the machine needs to boot up from hibernation.

So it seems this problem seems to occur when the machine was powered down (manual shutdown or by hibernation).

selimovd, 2021-04-09:
I have the same issue with the 2019 Macbook Pro.
Since AlDente 2.2 the battery is charging back to 100%. After waking up from sleep I have to kill AlDente and start it again, then it works again.

Do you need more information to solve this issue?

Thanks and best regards

MatthiasKerbl, 2021-04-09:
> I have the same issue with the 2019 Macbook Pro.
> Since AlDente 2.2 the battery is charging back to 100%. After waking up from sleep I have to kill AlDente and start it again, then it works again.
> 
> Do you need more information to solve this issue?
> 
> Thanks and best regards

Hi @selimovd,
do you have "Use Classic SMC key" enabled as well?

selimovd, 2021-04-09:
Hey @MatthiasKerbl ,

no, I do not have that activated. 
Only "Launch at Login" and "Allow Discharge (beta)".

MatthiasKerbl, 2021-04-09:
> Hey @MatthiasKerbl ,
> 
> no, I do not have that activated.
> Only "Launch at Login" and "Allow Discharge (beta)".

I see, please deactivate "Allow Discharge (beta)" since it still has some bugs. Afterward, you should restart AlDente and it would be nice if you could report back.

MatthiasKerbl, 2021-07-05:
Hi @ernst42,

We released a completely new version of AlDente yesterday which should fix the inconsistency you encountered.
You can find the new version here: https://github.com/davidwernhart/AlDente/releases

closed 2021-07-05 by MatthiasKerbl

ernst42, 2021-11-15:
Thanks, I will be able to test once I update my Mac to Big Sur or Monterey