AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #230: [feature request] Disable battery charging until it drops to certain value

Creeperface01, 2021-04-03:
Especially during lockdown this feature might be handy. I use my laptop approximately for 10 hours a day so it's at least one recharge every day. With this feature I could charge it once and keep the power connected without constantly charging the battery whenever it drops under the specified percentage. I think this feature would help me the most.

Not sure whether it's possible but according to existing beta discharge feature this should be pretty similar except I don't want to force discharge the battery but just leave it as it is.

Also this could be a premium feature I guess, if this would work I'll definitely consider purchasing it.

Creeperface01, 2021-04-03:
Or how does it work now? Is there any threshold to start charging the battery?

danfoerster, 2021-04-04:
For now, what you can do is, disable 'Allow Discharge', set your desired SoC and when it has reached that, put in a lower value in AlDente so it prevents to charge your battery until that lower threshold is reached. I have my Mac charge to around 70%, then change the value to 30 and the battery is not being charged until it has run down to 30% (which, if you don't use your Mac while its unplugged, could take months to do so)

Creeperface01, 2021-04-04:
Yeah that's exactly what I did. Just somehow didn't realise earlier, anyway still could be a nice feature.

MatthiasKerbl, 2021-07-05:
Hi @Creeperface01,

thank you for this great idea. We implemented exactly this feature in our Pro version of AlDente and it is called Sailing Mode.
You can read more about it on our website: https://apphousekitchen.com/

closed 2021-07-05 by MatthiasKerbl