AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #233: My MacBook Air M1 8/7 8GB RAM heats up after installing the app.

sohate, 2021-04-04:
I set the limit for 65%. And now when the charge reaches the limit (or during the charging) my MacBook heats up. Anybody else notice that?

MatthiasKerbl, 2021-04-04:
Hi @sohate,
which temperature and how much does it increase while charging and while staying at a certain percentage?

LeshawnG, 2021-04-07:
If this is useful information, I use TG Pro for my temp monitoring and my current ambient room temp is 28C with my M1 CPU and battery avg. temps are sitting constantly at 34C and my power manager dies are around 41C.
I'll check my temps again later when my ambient temp is lower around 21C. Also it should be noted that my Mac has been sitting on my lap for some time. 
I don't have these issues myself, I'm just interested with the temperature information.
[image]

sohate, 2021-04-09:
I think it was a mistake by me. The reason of heating up is Google Chrome: it was closed but was still eating my MacBook.

MatthiasKerbl, 2021-04-09:
> I think it was a mistake by me. The reason of heating up is Google Chrome: it was closed but was still eating my MacBook.

Hi @sohate,
thank you for clarifying. I think we can close this issue now.