AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #236: It is not working on MacBook Air M1

rharangozo, 2021-04-05:
AlDente keeps asking my password to start helper but even after that, it does not work as expected. Sometimes, the slider jumps back to 0. If i reinstall helper, i can set the desired percentage but it just simply does not work. 
If I run it with Rosetta 2, it works fine!

Hardware: Apple MacBook Air M1
OS: 11.2.3 (20D91)
AlDente: Version 2.2

Dhudong, 2021-04-06:
Same exact issue for me! How did you get to run it with Rosetta 2?

rharangozo, 2021-04-06:
> Same exact issue for me! How did you get to run it with Rosetta 2?

Navigate to /Applications in Finder, right click on AlDente, select 'Get Info' and check 'Open using Rosetta'

Dhudong, 2021-04-07:
> > Same exact issue for me! How did you get to run it with Rosetta 2?
> 
> Navigate to /Applications in Finder, right click on AlDente, select 'Get Info' and check 'Open using Rosetta'

Thanks man! Did that and it's working now. Hope this issue will be resolved soon. Thanks again.

rharangozo, 2021-04-15:
Now i have a more experience with AlDente on MacBook Air M1. Even if i run it with rosetta, it silently stops working, probably the helper crashes. If i reinstall it, sometimes it starts working again. Sometimes, i need to reboot. Unfortunately, it's unreliable.

AlfaMoski, 2021-04-27:
I am getting 
[image]
 message “com.davidwernhart.Helper” will damage your computer."

MatthiasKerbl, 2021-07-05:
Hi everyone,

We just released a new version of AlDente which should solve the problem you encountered.
You can download the new version here: https://github.com/davidwernhart/AlDente/releases
Sorry, it took so long but I hope you like the new version.

Kind regards,

Matthias

closed 2021-07-05 by MatthiasKerbl