AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #239: Have to do battery calibration less than one month

halqin, 2021-04-09:
Since my last issue [#207 ](url), I did full battery battery calibration at that time. But yesterday, only about 3 weeks later, I have this issue again. So I have to do battery calibration again! Which means the battery calibration should be done every three weeks? I doubt the effect on battery life span under such a recurrent deep exhaust and deep charge behavior.

MatthiasKerbl, 2021-04-09:
Hi @halqin,
thanks for reporting back again. What was your exact behavior the last three weeks? Did you just let it plugged in at a certain percentage or did you charge it a little bit? Normally, it should be enough to use it in battery mode just for a half cycle to avoid this issue. I never had this issue even after weeks of not charging my MacBook. However, I never used it for many months without charging at least once

halqin, 2021-04-09:
I set it charge to 60%. It should be a "safe" value. Sometimes I also use battery during these 3 weeks, in my living room, in my dining-table etc.. nothing special manipulate.

MatthiasKerbl, 2021-04-09:
This almost sounds like the exact same usage behavior I have with my MacBook but my battery calibration seems to be fine with one full cycle every 4-6 weeks.

closed 2021-04-11 by halqin