AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #240: Autodetect if running an intel or M1 Mac.

appetrosyan, 2021-04-09:
Installed today. Didn't work until I turned it use the classic SMC key. 

First the name is misleading, instead of (intel) being in parenthesis, you should actually have it 
`switch to intel SMC key (classic)`. 

Additionally, the end user shouldn't have to do this by hand.

MatthiasKerbl, 2021-04-09:
Hi @appetrosyan,

it was just recently that we found out about the new mode not working on every Intel MacBook. Which MacBook model do you have? We will definitely update this in a future update.

appetrosyan, 2021-04-09:
Ah! Thanks for getting back to me so soon. 

It's a 2017 Early 15'' MacBook Pro with Touch Bar.

MatthiasKerbl, 2021-04-09:
Interesting, I have a 2016 15" MacBook Pro with Touch Bar and it is working fine for me with Classic SMC Key disabled. Did you try it with Allow Discharge disabled?

appetrosyan, 2021-04-10:
I didn’t enable the `allow discharge`.

MatthiasKerbl, 2021-07-05:
Hi @appetrosyan,

our newest version of AlDente which was released yesterday detects if it’s an Intel or M1 MacBook on its own.

closed 2021-07-05 by MatthiasKerbl