AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #241: [Feature Request] Calendar check - disable Al Dente on specific events...

eighteentee, 2021-04-10:
How about a feature that could disable Al Dente if you have a calendar event which you specifically set, say a meeting, where you will be away from power for a significant amount of time. This could allow Al Dente to turn off the charge limit beforehand and charge the laptop to full capacity, ready to take on the road, away from AC.

This is something I have to do manually right now - I have to remember when I need the laptop on full charge and remember to disable Al Dente to do this.

MatthiasKerbl, 2021-04-16:
Hi @eighteentee,

this is a great feature idea. We will think about it but implementing a good charging schedule isn't easy.

Matthias

JamesMatt, 2021-07-06:
macOS Monterey support Shortcuts app, maybe you can think about how to integrate?

MatthiasKerbl, 2023-03-12:
Hi everyone,

We have just released AlDente 1.21.1 which has a new beta feature called Schedule. This feature allows users to set specific Actions in AlDente to run automatically on predetermined days, times, and intervals. For example, Calibration Mode can be set to run every two weeks.
You can read more about the Schedule feature and its current limitations here: https://apphousekitchen.com/feature-explanation-schedule/
You can find a full list of the Release Notes here: https://apphousekitchen.com/changelog/

Cheers,
Matthias

MatthiasKerbl, 2026-10-08:
Hi @eighteentee,

Since this request, we have added the Schedule feature, which allows you to automatically run Top Up, Calibration Mode, Discharge and more at specific days, times, and intervals. Additionally, AlDente supports Apple Shortcuts, so you can also trigger actions like Top Up through calendar-based automations in the Shortcuts app.
You can read more about it on our blog: [Feature Explanation: Schedule](https://apphousekitchen.com/feature-explanation-schedule/) and [Feature Explanation: Shortcuts](https://apphousekitchen.com/feature-explanation-shortcuts/)

Therefore, I will close this issue for now. If you think something is still missing, just let me know.

Best,
Matthias

closed 2026-10-08 by MatthiasKerbl