AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #242: macos bigsur  Not available

sunhongfan, 2021-04-10:
Hi, I like this software very much. It worked well before, but after I reinstalled the system and installed it again, it didn’t work properly. My battery level was 100 during installation. After installation, I restarted the computer, connected the charger all the time, and set the maximum value. At 60, it is always in a lightning state.

18macbookpro touchbar . bigsur

sunhongfan, 2021-04-10:
help me !!!!

MatthiasKerbl, 2021-04-16:
Hi @sunhongfan,

does AlDente still not work on your MacBook?

Matthias

sunhongfan, 2021-04-18:
> 嗨@sunhongfan，
> 
> AlDente仍无法在MacBook上运行吗？
> 
> 马蒂亚斯

It works, but when I get to 100, it doesn't work properly. It keeps charging

MatthiasKerbl, 2021-04-18:
> > 嗨@sunhongfan，
> > AlDente仍无法在MacBook上运行吗？
> > 马蒂亚斯
> 
> It works, but when I get to 100, it doesn't work properly. It keeps charging

I am not sure if I understand your issue correctly. Could you explain it in more detail?

sunhongfan, 2021-04-21:
> 详细

When the charger is plugged in and the power is 100%, it does not matter at this time, although I set the maximum value to 80, when I unplug the power and consume a certain amount of power, it will work normally.          18Macos bigsur

MatthiasKerbl, 2021-04-21:
> > 详细
> 
> When the charger is plugged in and the power is 100%, it does not matter at this time, although I set the maximum value to 80, when I unplug the power and consume a certain amount of power, it will work normally. 18Macos bigsur

When your MacBook is at 100% you could use the beta feature "Allow Discharge" to get it down to your desired charge limit. Otherwise, you can just unplug your MacBook and drain the battery to your charge limit and plug it in again afterward.

closed 2021-07-05 by MatthiasKerbl