AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #243: any idea how to compile this app for windows?

madkep, 2021-04-11:
i´m using my mac with windows bootcamp I prefer to use it this way, but the battery is very damaged by not having a limiter.

lockieluke, 2022-06-29:
it would be quite difficult to recompile the program for Windows as the program itself is written in Swift which is a programming language mostly for Apple devices(well it supports Windows and Linux technically but it's very limited)

MatthiasKerbl, 2022-11-09:
Hi @madkep,

If you have an Intel MacBook without Thunderbolt 3, the set Charge Limit in AlDente in macOS will automatically persist in Windows.
If you have an Intel MacBook with Thunderbolt 3, activate the Intel Mode in AlDente in macOS. Afterward, the Charge Limit will also persist in Windows.
You can read more about this topic on our blog:
https://apphousekitchen.com/feature-explanation-intel-mode/
https://apphousekitchen.com/different-categories-of-macbook-models/
https://apphousekitchen.com/adding-aldente-pro-support-to-16-new-intel-macbook-models-full-list-below/

Cheers,
Matthias

martin-braun, 2023-07-19:
@MatthiasKerbl I wonder if it was possible to update the charge limit from Windows or Linux without rebooting to MacOS. I've set the charge limit in Intel Mode, but once I reach the limit I would love to set it down a bit, so my battery doesn't constantly initiate another recharge when I pull the cable for just a brief second.

Is there any low level bridge to just update the charge limit for intel mode (T2 chip) from terminal?