AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #247: Helper file considered dangerous

nocom, 2021-04-13:
MBP 2016 15" Big Sur
I. get a warning immediately after installing that the. helper file. can damage my comput
[image]

iphoneio, 2021-04-13:
Same Here.
Apple say ""com.davidwernhart.Helper" will damage your computer.
Report malware to Apple to protect other users"😢
MacBook Pro (M1, 13-inch, 2020) macOS 11.2 Big Sur

[image]

MatthiasKerbl, 2021-04-13:
> MBP 2016 15" Big Sur
> I. get a warning immediately after installing that the. helper file. can damage my comput
> [image]

Hi @nocom,
Which version of AlDente do you use?

MatthiasKerbl, 2021-04-13:
> Same Here.
> Apple say ""com.davidwernhart.Helper" will damage your computer.
> Report malware to Apple to protect other users"😢
> MacBook Pro (M1, 13-inch, 2020) macOS 11.2 Big Sur
> 
> [image]

Hi @iphoneio,
which version of AlDente do you use?

nocom, 2021-04-13:
The free version since both other options are yellow and do not reactAm 13.04.21 um 09:16 schrieb Matthias Kerbl
                
            
            
                
                    Von: "Matthias Kerbl" ***@***.***>Datum: 13. April 2021An: "davidwernhart/AlDente" ***@***.***>Cc: "Mention" ***@***.***>,"nocom" ***@***.***>Betreff: Re: [davidwernhart/AlDente] Helper file considered dangerous (#247)
                
                

MBP 2016 15" Big Sur
I. get a warning immediately after installing that the. helper file. can damage my comput

Hi @nocom,
Which version of AlDente do you use?

[
{
"target": "https://github.com/davidwernhart/AlDente/issues/247#issuecomment-818502585",
"url": "https://github.com/davidwernhart/AlDente/issues/247#issuecomment-818502585",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

MatthiasKerbl, 2021-04-13:
Did you have any previous versions of AlDente installed?

nocom, 2021-04-13:
Yes but when I updated my Cocktail app the same warning popped up and I deleted AIDente. Am 13.04.21 um 16:00 schrieb Matthias Kerbl
                
            
            
                
                    Von: "Matthias Kerbl" ***@***.***>Datum: 13. April 2021An: "davidwernhart/AlDente" ***@***.***>Cc: "Mention" ***@***.***>,"nocom" ***@***.***>Betreff: Re: [davidwernhart/AlDente] Helper file considered dangerous (#247)
                
                

The free version since both other options are yellow and do not reactAm 13.04.21 um 09:16 schrieb Matthias Kerbl Von: "Matthias Kerbl" @.>Datum: 13. April 2021An: "davidwernhart/AlDente" @.>Cc: "Mention" @.>,"nocom" @.>Betreff: Re: [davidwernhart/AlDente] Helper file considered dangerous (#247) MBP 2016 15" Big Sur I. get a warning immediately after installing that the. helper file. can damage my comput Hi @nocom, Which version of AlDente do you use?

Did you have any previous versions of AlDente installed?

[

{

"target": "https://github.com/davidwernhart/AlDente/issues/247#issuecomment-818759551",

"url": "https://github.com/davidwernhart/AlDente/issues/247#issuecomment-818759551",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"name": "GitHub",

"url": "https://github.com"

}

}

]

MatthiasKerbl, 2021-04-16:
Hi,

try the following steps:

Quit and delete AlDente and go in Finder to Macintosh HD/Library/LaunchDaemons and delete “com.davidwernhart.Helper.plist”. Afterward, go to Macintosh HD/Library/PrivilegedHelperTools and delete “com.davidwernhart.Helper”. Restart your MacBook and download and install the newest version of AlDente again.

mrsbpier, 2021-05-11:
[image]

Same here...latest version of AlDente and macOS.

MatthiasKerbl, 2021-07-05:
Hey everyone,

we released a completely new version of AlDente yesterday which will fix the problems you encounter.
You can download the new version here: https://github.com/davidwernhart/AlDente/releases

Please uninstall the old version of AlDente before installing the new one.

closed 2021-07-05 by MatthiasKerbl