AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #248: [feature request] Charging schedule

tagir-gadelshin, 2021-04-14:
It would be great to be able to set some schedule. E.g. I know I’ll need to take my Mac with me in the morning. And I need the tool to charge it to 100% by the time I’ll leave, and this is on everyday basis

For the first version maybe at least changing the target value at certain time would be great. So I’ll put 100% to 6 a.m. and then my Mac will charge from 60 to 100 within a couple of hours.

predicting when to start charging will be great addition to that (so I set that I need 100 at 8 a.m. and the tool decides when to start charging based on history or some hard-coded data), but not necessary for the first version.

I’d pay for pro if it will has this feature

tosbsas, 2021-04-15:
Love That

MatthiasKerbl, 2021-04-16:
Hi @tagir-gadelshin,

thank you for your suggestions. We will discuss them but it is not easy to implement a good charging schedule.

Matthias

matyaska, 2021-08-02:
In addition to a charging schedule, it could be useful to charge automatically in conjunction with meetings. This is a feature the apple "optimize battery life" has. I understand that it is not trivial.

gusost, 2021-08-31:
Hey @MatthiasKerbl!
I just bought the Pro version, mainly because I like the whole idea to improve on battery life time.
I rarely use my MBA on battery power for long enough to warrant a full charge. 
But when I do I sometimes forget to set battery level to 100% in time.

An easier implementation of this feature request would be to have either a timer option like "In 2 hours set desired charge level to 100%" type thing. 
Could also be a one time-time thing: "At 4:00 set desired charge level to 100%"
However, as a developer, I know that any time you involve specific times / dates there is bound to be problems.
For this real simple, a single time input, that would include problems with AM/PM or 24h clock, what if the user changes timezones? Go by the one where it was set or the new local one etc, etc.

That being said. I think the feature with a timer would still be very useful. For me I always think of it earlier in the day but forget before going to bed.

Thanks for a great little app!

MatthiasKerbl, 2021-09-19:
Hi @gusost,

I am glad you like our little app!

Your idea regarding a one-time charging timer might be a simple but effective way to implement a first iteration of a charging schedule.

Cheers,
Matthias

colinjmatt, 2022-02-13:
> An easier implementation of this feature request would be to have either a timer option like "In 2 hours set desired charge level to 100%" type thing. Could also be a one time-time thing: "At 4:00 set desired charge level to 100%" 

I think many would see value in this, but I'd like to add to this particular suggestion and have a weekly schedule that can be configured by the user.

E.g., Sunday-Saturday, each day has a time you can specify to automatically start and stop the set charging limit, with a tick box for each day to enable/disable it.

Here's a visual representation of my example:
[image]

skrivanos, 2022-04-26:
+1

Something to just schedule a one-time "charge to 100% at 8am" would help me a ton.

DamonHD, 2022-08-30:
A thought: if you want to add some grid-smartness that could be "charge to 100% by 8am".  Not that it makes a huge difference but it might be more helpful anyway to you!

Rgds

Damon

MatthiasKerbl, 2023-03-12:
Hi everyone,

We have just released AlDente 1.21.1 which has a new beta feature called Schedule. This feature allows users to set specific Actions in AlDente to run automatically on predetermined days, times, and intervals. For example, Calibration Mode can be set to run every two weeks.
You can read more about the Schedule feature and its current limitations here: https://apphousekitchen.com/feature-explanation-schedule/
You can find a full list of the Release Notes here: https://apphousekitchen.com/changelog/

Cheers,
Matthias

tosbsas, 2023-03-12:
Will That bother Standby? And what Happens when machine is Turned on again ?______________________________Ruben GutknechtLima, Perú  🇵🇪 Cel: +51-99349-1515Voice/Fax: 49 (0) ***@***.*** 12/03/

MatthiasKerbl, 2023-03-12:
Hi @tosbsas,

Please take a look at our blog post here: https://apphousekitchen.com/feature-explanation-schedule/

Cheers,
Matthias

MatthiasKerbl, 2023-06-09:
Hi everyone,

Since our Schedule feature has officially been released for a while now, I will close this issue/feature request.

Best,
Matthias

closed 2023-06-09 by MatthiasKerbl