AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #249: Pro version questions

tosbsas, 2021-04-14:
Some questions:

When sleeping - stop charging - it says when activated its inhibiting charging - but that doesn't fit the description

I have found the UI confusing, especially Sailing Mode toggle. It appears that on is really off. Your comment applies to the Calibration Mode button as well.

It would be nice to know how long it takes before Sailing Mode kicks in and starts slowly draining your battery down to the specified level. That way we would know if it is working correctly or not.

So far I haven't seen any sailing

mikebore, 2021-04-14:
I clicked "Get Pro" in settings in the app months ago, but have not heard anything from Al Dente. As soon as I heard about the beta I sent an email requesting to be a beta tester, but have had no response.

Are the number of beta testers very limited, or have you had an overwhelming response? 

What was the purpose of the button :"Get Pro" in Settings? 

Thanks

tosbsas, 2021-04-14:
I think its a small team of Devs, they are doing their best to get an awesome app out. But communication is not their mayor skill :-)

davidwernhart, 2021-04-14:
Hey @tosbsas,

thank you for your feedback! :)

Sleeping -> stop charging means, that AlDente automatically stops charging as soon as it detects that the system is going to sleep. Your MacBook should not lose nor gain battery charge when sleeping and this setting is enabled.

Sailing mode also does not actively discharge your battery. It simply acts as a form of hysteresis, so your Macbook is not always starting to charge if it loses a few percent over time.
An example scenario would be: You are on 50%, have your charge limit on 70% and sailing mode set to 5%. After you reached 70%, AlDente will stop charging and, because of sailing mode, will only start charging again once the battery is down to 65%. When sailing mode is working, a small paper plane appears on the charging bar.

I hope that clears things up for you. You are right in that we still have to do some improvement regarding the explanations of the settings.

Of course, don't hesitate to ask if you have more questions or bug reports!

Best regards,

David & Matthias

davidwernhart, 2021-04-14:
Hey @mikebore,

if you registered your email on the "Get Pro" page, we for sure sent you an email regarding the beta testing program, but sadly some emails cannot be delivered because of some higher forces (technical issues, your email provider's spam policy, ...). Maybe our mail has landed in your spam folder, but either way, we are sorry for that.

Since we are only a team of two, we decided to keep the number of initial beta testers relatively low, so we can keep track of all the bugs and stay focused.

That being said, it definitely won't be long until we release the full version to everyone, and your early engagement will definitely be appreciated! :)

Thank you very much for your support,

David & Matthias

mikebore, 2021-04-14:
Thanks. Definitely not my spam or junk, and not in bin, so must be a higher force. I will follow this thread with interest and be a beta tester by proxy.

mikebore, 2021-04-14:
> Sailing mode also does not actively discharge your battery. It simply acts as a form of hysteresis, so your Macbook is not always starting to charge if it loses a few percent over time.
> An example [...]

Brilliant! this is similar to what was being discussed in this thread
[https://github.com/davidwernhart/AlDente/discussions/172](https://github.com/davidwernhart/AlDente/discussions/172)

In practice, at least on my system and usage, it will take a looong time to naturally get down from 70 to 65. I think I am doing sailing manually as discussed in that thread.

tosbsas, 2021-04-15:
In Start seeing sailing :-)

mikebore, 2021-04-15:
I have been playing with the options and believe I understand sailing which will be my standard usage.

I understand the two buttons "Discharge" and "Top up" are simple manual overrides to any percentage setting, and charge now or discharge now. 

VERY pleased to see that the "Discharge" option is disabled with a message in Clamshell mode. "Enable discharge" in version 2.2 was  nasty poo-trap for Clamshell users.

Heat Protection seems straightforward and explained by the pop out. Can't think of a reason not to have that on all the time.

Not explored calibration yet...I assume that will be a lengthy cycle of discharge and charge, and may not work in Clamshell mode.

AppleRobert, 2021-04-15:
Consider refining the action of the Calibration button to make it more clear when it has been turned on an when it has been turned off.

AppleRobert, 2021-04-15:
This beta has been very stable. Outside of some possible cosmetic touch up to the settings area, I haven't come across any bugs, yet.

tosbsas, 2021-04-15:
yep, second that

mikebore, 2021-04-16:
Very stable here too. 

The only slight puzzle is that I set it up 36 hrs ago with 70% and 5% sailing on. At that point the charge was 70%. By the end of yesterday the charge was 75% where it has remained. Under sailing it is going to take a very long time to naturally discharge to 70% which is fine. I think the reason it rose above 70% was during a couple of restarts where the Mac was plugged in and shut down for a short while, so would be charging. I guess it doesn't take long to increase charge on a shut down Mac from 70-75%. I am going to come out of Clamshell mode so that I can do a manual discharge to under 70% and observe what happens more closely, especially if I shut down.

mikebore, 2021-04-16:
> I understand the two buttons "Discharge" and "Top up" are simple manual overrides to any percentage setting, and charge now or discharge now.

What I wrote above is wrong. The "Discharge" button does not override the percentage setting but discharges down to the percentage setting.

mikebore, 2021-04-18:
> I think the reason it rose above 70% was during a couple of restarts where the Mac was plugged in and shut down for a short while.

Just confirmed this. I spent sometime booted form Recovery and from an Bootable USB installer, maybe 30 minutes. When I boot back to my main internal battery is at 97% and in sailing mode. All logical and expected. (I was just surprised how quickly it happened before). Time to use the discharge button.

MatthiasKerbl, 2021-05-03:
Hi guys,

thanks again for your support.

You can read more about the details of the features of AlDente Pro in our blog: https://apphousekitchen.com/blog/

Kind regards,

Matthias

closed 2021-05-03 by MatthiasKerbl