AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #250: Suggestion for inclusion in Pro Version

mikebore, 2021-04-15:
I expect many people who care enough about batteries will have installed some or all of Coconut, Battery Monitor 2, or Marcel Bresink's Battery Monitor. All of these utilities provide extra info extracted from the system by the ioreg command (_ioreg -w 0 -f -r -c AppleSmartBattery_). Particularly interesting to me is the current max capacity or Health %. None of them provide any control like AlDente.

How about incorporating some or all of this info into AlDente to make it the only, and all-can-do Battery app?

AppleRobert, 2021-04-15:
Excellent idea, Mike.

MatthiasKerbl, 2024-02-09:
Hi everyone,

We have just released AlDente 1.25 which includes a new widget for battery data. The new data widget can be activated in the stats category of in the settings. To get AlDente 1.25 you just need to update within the app under Settings/General/Check for Updates.

Best,
Matthias

closed 2024-02-09 by MatthiasKerbl