AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #251: M1 macbook air keyboard not working properly

m1mba, 2021-04-16:
Hello! I just noticed after installing the helper that my a and d keys are not working. 
I don´t know if it´s related so I ask you guys.
I´ve already tried uninstalling, wiping the ssd and reinstalling OS from scratch. 
Still not working. I´ve also tried an external keyboard and it does work.
I would also revive and restore but I don´t have another macbook to do so.
Any ideas?
Thank you!

m1mba, 2021-04-16:
At the same time I installed rosetta 2.

m1mba, 2021-04-17:
I have also disabled all keyboard shorcuts

MatthiasKerbl, 2021-04-17:
Hi @m1mba,

I am very sorry you have issues since you installed AlDente. However, I don't think AlDente has anything to do with your keys not working properly anymore. In my opinion, this could be a hardware issue.

m1mba, 2021-04-18:
Hello Matthias, thank you for your answer. I´m starting to think it is indeed a hardware problem. I wish I could try restoring the firmware. I guess I´ll take it to the store.

MatthiasKerbl, 2021-04-19:
I think we can close this issue for now, since it is probably not correlated with AlDente. However, if you find anything else, let us know.

closed 2021-04-19 by MatthiasKerbl