AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #252: Al Dente lets the battery charge above the limit if the commputer sleeps - M1

mzsombor86, 2021-04-17:
The problem is the following - the subject is an M1 Macbook Air:
If I plug in the charger at a lower SoC level, and then I close the lid, the computer charges to 100%.
If I don't close the lid, and keep using the computer, the charging stops at the setted level, and if I close the lid after reaching the limit, then everything is fine, the computer will keep the desired state of charge.

MatthiasKerbl, 2021-04-17:
Hi @mzsombor86,

AlDente can not control the Charging behavior of your MacBook while it is in sleep. Therefore, it will keep charging until it is at 100%.

koolman2, 2021-04-21:
My M1 Air and Pro both stop up to 10% above my target when closed. I have the option "Prevent computer from sleeping automatically when the display is off" enabled on both devices. It at least seems to stop them from going all the way to 100%.

MatthiasKerbl, 2021-04-22:
> My M1 Air and Pro both stop up to 10% above my target when closed. I have the option "Prevent computer from sleeping automatically when the display is off" enabled on both devices. It at least seems to stop them from going all the way to 100%.

Hi @koolman2,
which version of AlDente do you use? Do you have Apple's Optimised battery charging disabled?

koolman2, 2021-04-22:
Now that I think about it, I'm not sure I've ever charged the Pro with the lid closed, so let's ignore that machine for now.

The Air is running 2.1.2. It has Optimized Charging disabled. I'm updating it now to 2.2 (this is my partner's device who couldn't care less about this). It is routinely plugged in to charge while the lid is closed. I have the limit set to 70%, and it is usually found in the morning to have charged to anywhere from 70% to 82%. It is charged using the included 30 W charger.

Here are the batter/adaptor settings:

[image]
[image]

This machine was unplugged last night, so ignore the time of last full charge.

dmelgar, 2021-06-20:
By closing this issue, does it mean its a known issue that the battery will charge to 100% if it goes to sleep?
Sounds like the SMC settings are only effective for stopping charging, but that setting the charge limit only works when AlDente is actively running to be able to stop it. Maybe if it had stopped when it went to sleep it works.

I've been running AlDente on an Intel MacBook Pro and didn't have this issue. Is this unique to the M1? Maybe there isn't a charge limit SMC setting?

MatthiasKerbl, 2021-06-26:
> By closing this issue, does it mean its a known issue that the battery will charge to 100% if it goes to sleep?
> Sounds like the SMC settings are only effective for stopping charging, but that [...]

Hi @dmelgar,

AlDente works differently on Intel and M1 MacBooks.
On Intel MacBooks in Intel Mode, the MacBook will stop charging even if the MacBook is closed or turned off. On M1 MacBooks on the other hand, AlDente can only stop charging when the MacBook is turned on (not in sleep, not turned of, not closed). If you set the charging limit to 80% and you close the lid after the M1 MacBook has reached 80%, it will stay at 80%.

closed 2021-06-26 by MatthiasKerbl

hotsince99, 2021-11-13:
> Hi @dmelgar,
>
> AlDente works differently on Intel and M1 MacBooks. On Intel MacBooks in Intel Mode, the MacBook will stop charging even if the MacBook is closed or turned off. On M1 MacBooks on the [...]

Hi Matthias. My MacBook Pro is still charging when it's turned off. Is this counts as bug?

MatthiasKerbl, 2021-11-13:
> Hi Matthias. My MacBook Pro is still charging when it's turned off. Is this counts as bug?

Hi @hotsince99,

The following article explains in detail what happens when your MacBook is turned off.
https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/

Cheers,
Matthias

kevinwheeler, 2023-04-11:
> AlDente can not control the Charging behavior of your MacBook while it is in sleep. Therefore, it will keep charging until it is at 100%.

@MatthiasKerbl The link you posted disagrees with what you said. It says Al Dente can control sleep behavior.

From https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/ for Apple Silicon:

"The only way to avoid letting your MacBook go to 100% is to not shut it down and let your MacBook go to sleep instead or to unplug it before you turn it off."

Also, if it's the case that Al Dente can't control sleep charge behavior, is it still better for the longevity of your battery and your mac overall to prevent your mac from sleeping, so that the charge amount can stay constant?

MatthiasKerbl, 2023-04-12:
> > AlDente can not control the Charging behavior of your MacBook while it is in sleep. Therefore, it will keep charging until it is at 100%.
> 
> @MatthiasKerbl The link you posted disagrees with what you said. It says Al Dente can control sleep behavior.
> 
> From https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/ for Apple Silicon:
> 
> "The only way to avoid letting your MacBook go to 100% is to not shut it down and let your MacBook go to sleep instead or to unplug it before you turn it off."
> 
> Also, if it's the case that Al Dente can't control sleep charge behavior, is it still better for the longevity of your battery and your mac overall to disable/prevent your mac from sleeping, so that the charge amount can stay constant?

Hi @kevinwheeler,

If you have the feature "Stop charging when sleeping" enabled, your MacBook will not continue charging when it is asleep.

Please also take a look at the following explanations as well:

- https://apphousekitchen.com/feature-explanation-disable-sleep-until-charge-limit/
- https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/

> Also, if it's the case that Al Dente can't control sleep charge behavior, is it still better for the longevity of your battery and your mac overall to disable/prevent your mac from sleeping, so that the charge amount can stay constant?

These are the best to worst options in my opinion:

1. Turn off and unplug your MacBook at a healthy battery percentage when you are not using it for longer periods.
2. If you prefer to not turn off your MacBook because you like to have it wake up quickly when you use it, enable the features "Stop charging when sleeping" and "Disable sleep until charge limit".
3. Don't use AlDente.

Please keep in mind that more research and data are needed. This is just my personal opinion based on the research I read.

Cheers,
Matthias