AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #253: Battery level exceeds the limit

keremsirin, 2021-04-18:
I set max battery level to 80% and like for every 8-10 minutes battery level becomes 81% then it cuts energy and again turns 80% because of allowing discharge. It's exceeds the limit.

MatthiasKerbl, 2021-04-19:
Hi @keremsirin,
do you have Apple's Optimised Battery Charging disabled?

keremsirin, 2021-04-19:
Hi @MatthiasKerbl,
No I did not. Is it the problem?

MatthiasKerbl, 2021-04-19:
Yes, this behavior is due to Optimised battery charging being still enabled. Therefore, we recommend disabling it in our README.
You should disable Optimise battery charging and try AlDetne again.

keremsirin, 2021-04-19:
I'm sorry I did not see that. Thank you

closed 2021-04-19 by keremsirin

MatthiasKerbl, 2021-04-19:
> I'm sorry I did not see that. Thank you

No worries, you are welcome.