AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #254: AlDente seem to interfere with optimized charging

jacobedv, 2021-04-19:
After using AlDente flawless for a few days, my external Monitor gets black, when I close the lid of my MacBook Air M1. 
After disabling AlDente, I can close the lid and still have a working external monitor.
My macOS is 11.3b8, monitor is a benQ PD2710QC with integrated USB-C dock.
One annotation: I had the max. charge limited to 60% for several days, after disabling AlDente, battery showed 83% charge immediately, after a few minutes 87% and then it stopped loading with the message, that the charge will go up to 100% at 6p.m. (5 ½ hours from now). So it seems to be a problem with the optimized charge setting which was still enabled the whole time. Maybe AlDente should disable optimized charge by default. 
I will test AlDente with optimized charge disabled and will post the result in a few days.

MatthiasKerbl, 2021-04-19:
Hi @jacobedv,

did you have "Allow Discharge" enabled?
And yes, we definitely recommend disabling Optimised battery charging to avoid interference if AlDente.

jacobedv, 2021-04-20:
Ja, hab ich. Das hat auch funktioniert. 

Wenn ich Aldente mit 60% aktiviere und noch 80% im Akku habe, entlädt er im Betrieb bis zu den 60% und hält dann den Ladestand.
Nach dem Abschalten der Batterieoptimierung trat der Fehler mit dem „spontanen“ Schlafzustand bei Anschluss eines ext. Monitors nicht mehr auf.
Ich werde es noch ein paar Tage beobachten und dann auf Github meinen Kommentar ergänzen. 

Wenn das wirklich das Problem war, sollte Aldente die Deaktivierung der Apple Batterieoptimierung selbst vornehmen oder zumindest aktiv vorschlagen.

Viele Grüße, Johannes Jacob

	Jacob EDV
        Service und Support für Apple Computer seit 1988

        Frankfurter Landstraße 116
        64291 Darmstadt (Arheilgen)

        Tel.:   		0 6151 899 899
        Mail:		***@***.***

        http://www.jacobedv.de/ <http://www.jacobedv.de/>

	Anfahrt: https://g.page/jacob-edv?gm

	Öffnungszeiten:
	Mo., Di., Do. und Fr. 10-12 und 15-18 Uhr
	und nach Absprache.
	Am Mittwoch hat die Werkstatt geschlossen.

MatthiasKerbl, 2021-04-20:
> Ja, hab ich. Das hat auch funktioniert. Wenn ich Aldente mit 60% aktiviere und noch 80% im Akku habe, entlädt er im Betrieb bis zu den 60% und hält dann den Ladestand. Nach dem Abschalten der [...]

Das ist gut zu hören, wir haben in unseren README stehen, dass Optimised battery charging deaktiviert werden sollte. Aber natürlich wäre es noch besser wenn es gleich bei der installation vorgeschlagen wird.

LG

closed 2021-04-20 by MatthiasKerbl