AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #255: Confusion about Pro beta version "Allow Charging" option when sleeping.

mikebore, 2021-04-20:
My interpretation by just looking at the settings as in the screenshot, is that with the slider in the **left** default position charging is **not** allowed and I would have to turn it ON by moving to the **right** position to allow. I would have to turn the setting ON by moving it to the right to allow charging.

Looking at the dropdown that appears when hovering it seems to be the opposite?

[image]

Also I don't understand why the statement says that with charging inhibited you may overshoot desired percentage? Doesn't "Charging inhibited" mean no charging so how will it overshoot?

Does anybody else find this confusing?

MatthiasKerbl, 2021-04-20:
Hi @mikebore,

this is a little bit confusing right now and we already fixed it in our final version.
For now, the text next to the slider is the current state.

AlDente can not control the charging of your MacBook when your MacBook is asleep. Therefore, the MacBook will keep the charging state it had before going to sleep. When your MacBook is currently charging to your desired limit and your MacBook is going into sleep, it will overshoot your limit and charge to 100% since AlDente is not active while your MacBook is in sleep.

Hope that helps.

mikebore, 2021-04-20:
Thanks very much for clarifying.

closed 2021-04-21 by MatthiasKerbl